autowarefoundation/autoware_universe #7255: Street light is not perceived as an obstacle.

Kim-mins, 2024-06-04:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi team,

I'm currently running Autoware with Carla, and I found a situation that a street light is not properly perceived as an obstacle and the vehicle eventually collides with it.
Here are a front view and rviz video: [[rviz](https://drive.google.com/file/d/15IiPQJzrXU3OTtIbkjzMJyRpVn5SuJXJ/view?usp=sharing)] [[frontview](https://drive.google.com/file/d/1XbufX85yv3Tamy3uE1P7kr8ejoWolAry/view?usp=sharing)]

At 0:12 of the rviz video, the ego vehicle invades the sidewalk, and it eventually collides with a street light while recovering from the error.
I know that the bridge I used had an issue on processing control command so that is not the point, but while driving out from the sidewalk, I think that the vehicle should perceive the street light as an obstacle and stop before the collision.

### Expected behavior

I hope the vehicle to perceive the obstacle properly, and stop before any collision.

### Actual behavior

However, the vehicle collides with the obstacle.

### Steps to reproduce

Here's the ros2bag file for the reproduction: [[ros2bag](https://drive.google.com/drive/folders/1dobX9eNsJLCSZcGdDN-ZSRlWmFWX9JV_?usp=sharing)]

### Versions

- autoware.universe: [7877192cb5700c1db15109dcd959785fc381951d](https://github.com/autowarefoundation/autoware.universe/tree/7877192cb5700c1db15109dcd959785fc381951d)
- Carla: 0.9.15

sgNicola, 2024-06-04:
Hi~,I also encountered this problem and I think this is due to that the vector map used in Carla doesn't contain the traffic lights.

Kim-mins, 2024-06-04:
Hi @sgNicola

I also aware that our carla map does not contain traffic light information.
My point on this issue is not the traffic light, but the obstacle at front(in this case, street light), which is not properly perceived as an obstacle.

sgNicola, 2024-06-04:
I see, thank you for your explanation, the route is strange.

maxime-clem, 2024-06-13:
I confirmed that AEB (https://autowarefoundation.github.io/autoware.universe/main/control/autoware_autonomous_emergency_braking/) works as expected with your bag.
This means we can detect the collision, but because of the incorrect control, the collision detection is very late and even if we trigger an emergency stop it is too late to avoid the collision.
By default even if AEB detects a collision the emergency stops is not activated. See this discussion if you want to enable it: https://github.com/orgs/autowarefoundation/discussions/4669.

maxime-clem, 2024-06-13:
> By default even if AEB detects a collision the emergency stops is not activated. See this discussion if you want to enable it: https://github.com/orgs/autowarefoundation/discussions/4669.

In the default configuration there is no emergency stop with AEB and no messages in the logs. (see  https://github.com/orgs/autowarefoundation/discussions/4669)

I confirmed on the bag that AEB works as expected locally with the current Autoware. However in your bag the output of AEB never switches to the emergency mode and I could not understand why. If you encounter another similar issue please feel free to create another issue.

Otherwise I think we can close the present issue since the street light is correctly perceived as an obstacle by AEB with the current Autoware.

Kim-mins, 2024-06-13:
Thank you for the response @maxime-clem!

I think I got your point. thank you for the details.
I'll close this issue soon, and I'll let you know if I encounter the similar issue.

Thanks!

closed 2024-06-13 by Kim-mins