autowarefoundation/autoware_universe #7272: Error route planning in bidirectional lanes

NorahXiong, 2024-06-05:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

A new bug is detected when fixing #6803. An unexpected route is produced when setting the start and goal in bidirectional lanelets as specified in **Expected behavior** section and **Actual behavior** section.

Set start and goal in the [reference map](https://github.com/user-attachments/files/15573521/overlap11.osm.txt) as below: 
[image]

Planning behaviors:
[Overlap_Bug_New.webm](https://github.com/autowarefoundation/autoware.universe/assets/103234047/9903e319-2697-408d-a12b-8cc20141b44c)

### Expected behavior

A drivable or no route should be generated rather than an incorrect route.
[image]
[image]

### Actual behavior

An unexpected route is produced.
[image]

### Steps to reproduce

See **Description** section.

### Versions

- OS: docker
- ROS2: humble
- Autoware: up to 20240530

takayuki5168, 2024-07-03:
@NorahXiong cc @soblin @kosuke55 
The planning in the bidirectional lanes is not supported for now.

NorahXiong, 2024-07-05:
> @NorahXiong cc @soblin @kosuke55 The planning in the bidirectional lanes is not supported for now.

@takayuki5168 Then I will close the issue. Any other comments?

closed 2024-07-10 by NorahXiong