autowarefoundation/autoware_universe #7415: Compiling Autoware

YuqiHuai, 2024-06-10:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I am attempting to install Autoware following source installation method but always fail during `colcon build`.

### Expected behavior

Autoware builds without error.

### Actual behavior

```
--- stderr: yabloc_monitor                                                                                                                   
/home/yuqi/autoware/src/universe/autoware.universe/localization/yabloc/yabloc_monitor/src/yabloc_monitor_core.cpp: In constructor ‘YabLocMonitor::YabLocMonitor()’:
/home/yuqi/autoware/src/universe/autoware.universe/localization/yabloc/yabloc_monitor/src/yabloc_monitor_core.cpp:27:39: error: literal operator suffixes not preceded by ‘_’ are reserved for future standardization [-Werror=literal-suffix]
   using std::chrono_literals::operator""ms;
                                       ^~~~
cc1plus: all warnings being treated as errors
gmake[2]: *** [CMakeFiles/yabloc_monitor.dir/build.make:90: CMakeFiles/yabloc_monitor.dir/src/yabloc_monitor_core.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/yabloc_monitor.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< yabloc_monitor [10.3s, exited with code 2]
Aborted  <<< eagleye_rt [6.91s]                                                                                      
Aborted  <<< mrm_comfortable_stop_operator [7.05s]                                                                   
Aborted  <<< autoware_external_api_msgs [6.79s]
Aborted  <<< automatic_pose_initializer [6.93s]                                                                      
Aborted  <<< qp_interface [5.35s]                                                                                    
Aborted  <<< mrm_emergency_stop_operator [6.94s]                                                                     
Aborted  <<< tier4_hmi_msgs [7.81s]
Aborted  <<< tier4_perception_msgs [10.8s]
Aborted  <<< morai_msgs [14.4s]                                                                                      
Aborted  <<< ad_api_adaptors [9.12s]                                                                                              
Aborted  <<< autoware_auto_vehicle_msgs [13.4s]
Aborted  <<< traffic_light_arbiter [11.0s]                                                                                           
Aborted  <<< livox_tag_filter [12.6s]                                                                                                
Aborted  <<< duplicated_node_checker [13.9s]                                                                                         
Aborted  <<< polar_grid [9.23s]                                                                                                     
Aborted  <<< diagnostic_graph_aggregator [14.8s]                                                                                    
Aborted  <<< radar_scan_to_pointcloud2 [8.49s]                                                                               
Aborted  <<< dummy_infrastructure [18.0s]                                                                                     
Aborted  <<< eagleye_fix2kml [19.0s]                                                                                     
Aborted  <<< map_tf_generator [27.0s]                                                                                 
Aborted  <<< yabloc_common [30.9s]                                                                                    
Aborted  <<< costmap_generator [35.2s]                                                                                 
Aborted  <<< nebula_ros [1min 26s]                                             

Summary: 104 packages finished [1min 30s]
  1 package failed: yabloc_monitor
  23 packages aborted: ad_api_adaptors automatic_pose_initializer autoware_auto_vehicle_msgs autoware_external_api_msgs costmap_generator diagnostic_graph_aggregator dummy_infrastructure duplicated_node_checker eagleye_fix2kml eagleye_rt livox_tag_filter map_tf_generator morai_msgs mrm_comfortable_stop_operator mrm_emergency_stop_operator nebula_ros polar_grid qp_interface radar_scan_to_pointcloud2 tier4_hmi_msgs tier4_perception_msgs traffic_light_arbiter yabloc_common
  6 packages had stderr output: diagnostic_graph_aggregator livox_tag_filter nebula_ros traffic_light_arbiter yabloc_common yabloc_monitor
  206 packages not processed
```

### Steps to reproduce

Followed steps on https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/source-installation/

### Versions

- Autoware release v1.0

KYabuuchi, 2024-06-11:
@YuqiHuai Thank you for reporting the issue. However, in my environment, the error does not occur, and the build succeeds.
Could you please provide your OS and PC architecture to help identify the cause?

YuqiHuai, 2024-06-11:
Hi @KYabuuchi thanks for getting back to me! I noticed build would succeed in the docker container but for some reason fails outside.

I am currently running Intel Core i9-12900K running Ubuntu 22.04.1 (Linux yuqi-MS-7D31 6.5.0-35-generic #35~22.04.1-Ubuntu), cmake version 3.22.1

I am currently trying to develop a Python script that subscribes to different Autoware topics and I wanted to develop this outside of the docker container therefore I attempted to build from source. Considering my purpose and this build issue potentially being specific to myself, would you recommend I stay with using the container image for now?

KYabuuchi, 2024-06-11:
@YuqiHuai  Thanks for sharing the OS and other info. However, I don't see anything that could be the cause...
I don't know of any solutions right now, so I would recommend using the environment once you are able to compile it. Sorry.

KYabuuchi, 2024-06-11:
I noticed that literals become warnings in g++7 and earlier. Are you using older g++ in your environment?

[image]

[image]

reference: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=65923#c8

YuqiHuai, 2024-06-11:
Hi @KYabuuchi , thanks for helping me to do more investigation. Apparently I have 
```
gcc --version
gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
```
therefore it might be something else :(

KYabuuchi, 2024-06-11:
Hmm... :cry: I can't provide an immediate solution. Sorry

knzo25, 2024-06-11:
Sorry for barging in. could you check if you have any non standard environment variables set with  `printenv`?
As a hotfix, in yout environment you could probably use `--cmake-args` and `-Wno-error=literal-suffix`

YuqiHuai, 2024-06-11:
@KYabuuchi no worries! I guess if this is really just a specific issue on my machine we can close this issue, I should try different ways to figure this myself and maybe provide an update if I am able to figure it out. Currently I will stay with docker installation.

@knzo25 Thank you for barging in! Well, unfortunately I am not too familiary with cmake args and I am not sure what "non standard environment" there could be. I have already gone through many problems because I was using `pyenv`, I had to totally disable that and use only system Python. This hot fix will likely work and thank you for this suggestion! But for now I am deciding to stay with docker build as I am able to very consistently build Autoware in the container.

KYabuuchi, 2024-06-12:
Since this seems to be the author-specific issue, I will close the issue. 
If anyone has new information, or finds that it is an autoware issue, please feel free to reopen it. :pray:

closed 2024-06-12 by KYabuuchi