autowarefoundation/autoware_universe #7427: Potential output topic mismatch

YuqiHuai, 2024-06-11:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I am studying the source code of Autoware and noticed the input/output topics are often remapped in launch files. I am expecting each node should register publishers for topics that I can find in the launch file.

### Expected behavior

Topic name should match in implementation and in launch file remap.

### Actual behavior

In [autoware behavior velocity planner node implementation](https://github.com/autowarefoundation/autoware.universe/blob/main/planning/autoware_behavior_velocity_planner/src/node.cpp), we have
```c++
  stop_reason_diag_pub_ =
    this->create_publisher<diagnostic_msgs::msg::DiagnosticStatus>("~/output/stop_reason", 1);
```
but in [launch file](https://github.com/autowarefoundation/autoware.universe/blob/main/planning/autoware_behavior_velocity_planner/launch/behavior_velocity_planner.launch.xml) we have
```xml
<remap from="~/output/stop_reasons" to="/planning/scenario_planning/status/stop_reasons"/>
```

Is this mismatch intended?

### Steps to reproduce

Reading source code of node implementation and launch file.

maxime-clem, 2024-06-14:
You are right this is a bug.
However that launch file is not used by default so it should not cause any issue when running Autoware.
The `obstacle_stop_planner` is launched in the motion planning launch file (if selected with the `motion_stop_planner_type`): https://github.com/autowarefoundation/autoware.universe/blob/main/launch/tier4_planning_launch/launch/scenario_planning/lane_driving/motion_planning/motion_planning.launch.xml#L188

Some example launch files are provided in the individual packages, but they are not the ones used when running the default Autoware setup.

maxime-clem, 2024-06-19:
I am closing this since there is no bug at runtime.
We could discuss removing the unused  launch files as they can cause confusion if they are not constantly updated.

closed 2024-06-19 by maxime-clem