autowarefoundation/autoware_universe #7453: LOAM Based Localization Task 5 - Implement Localization

ataparlar, 2024-06-11:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

The localization output is needed after all the pipeline.

### Purpose

Finding the optimal position for ego vehicle

### Possible approaches

- After finding the transform between point cloud frame and LOAM edge map, go backwards from LiDAR frame to `base_link`. Transform them with the transformation between clouds.

### Definition of done

- [ ] Give an output position
- [ ] Give an output orientation
- [ ] Give an accuracy of calculated optimal position if possible
- [ ] Compare it with NDT pose or some other pose sources.

yukkysaito, 2024-06-20:
Thank you for the suggestion. However, according to the community guidelines, issues are meant for reporting bugs.so please repost it in the GitHub discussion.
https://autowarefoundation.github.io/autoware-documentation/main/support/support-guidelines/

xmfcx, 2024-10-08:
- See https://github.com/autowarefoundation/autoware.universe/issues/7454#issuecomment-2399877539 for the final status.

Closing this since there are no resources to complete this task for now.

If anyone is interested, feel free to reopen.

closed 2024-10-08 by xmfcx