autowarefoundation/autoware_universe #7563: Misdetects the pedestrian near a car

sgNicola, 2024-06-18:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

The perception system fails to detect and track the pedestrian when they are close to the car. It only recognizes the pedestrian once they move away from the vehicle.
[image]
[image]
https://github.com/autowarefoundation/autoware.universe/assets/115910640/b538ca29-b0c7-4f02-a50a-64bf7100e8c2

### Expected behavior

Detect the pedestrian correctly

### Actual behavior

Miss detects the pedestrian near the car.

### Steps to reproduce

1. cd autoware/src/sensor_kit/external/awsim_sensor_kit_launch/awsim_sensor_kit_launch/launch，modify the awsim_sensor_kit, only one top lidar is used.
[image]
2. ros2 launch autoware_launch logging_simulator.launch.xml map_path:=map_path vehicle_model:=sample_vehicle sensor_model:=awsim_sensor_kit
3. replay this bag [https://drive.google.com/file/d/1FFE1XTrY-A8ELE1zi32bvGfe-Hah4u7P/view?usp=sharing](url)

### Versions

-OS: Ubuntu 22.04
-ROS2: Humble
-Autoware: Autoware.universe main

### Possible causes

The point clouds of pedestrians and cars are connected together， and not recognize correctly

sgNicola, 2024-06-20:
Here is the video of the rviz2 [https://github.com/autowarefoundation/autoware.universe/assets/115910640/b538ca29-b0c7-4f02-a50a-64bf7100e8c2](url)

badai-nguyen, 2024-07-04:
@sgNicola Thank you for your issue report. 
I think the current DNN model (Centerpoint) is not good enough for pedestrians nearby the vehicle, especially for the large vehicle like bus because of lack of training dataset. 
Could you try to enable `detection_by_tracker` for pedestrians by [this](https://github.com/autowarefoundation/autoware_launch/blob/000527bde7b1a4e4d643591d031818169802188d/autoware_launch/config/perception/object_recognition/detection/detection_by_tracker/detection_by_tracker.param.yaml#L10) and check this again?

amadeuszsz, 2024-07-08:
Out of curiosity, I tried with TransFusion, hoping intensity field will play main role here. Despite the fact the pedestrian and car results with different intensities, it still brings the same result as CenterPoint.
[image]

For TransFusion It could be due to different intensity profile in input data. I see Velodyne here, AFAIK dataset consist of Velodyne data as well. Therefore, the source for both ML models is lack of similar samples in dataset as @badai-nguyen mentioned.

amadeuszsz, 2024-10-17:
Closing this issue due to inactivity. Please feel free to reopen it if necessary.

closed 2024-10-17 by amadeuszsz