autowarefoundation/autoware_universe #7598: Support ROS 2 Jazzy Jalisco

HansRobo, 2024-06-20:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

The new distro, Jazzy Jalisco is out!
https://discourse.ros.org/t/ros-2-jazzy-jalisco-released/37862

Lets's start planning to support it.

### Purpose

The purpose is to support new distro, Jazzy Jalisco.
It is inevitable that some users will request Jazzy support in the near future.
In the process of supporting Jazzy, we will also see positive results for Autoware running on Humble, such as removing deprecated features.

### Possible approaches

- Identify and migrate features that are deprecated in humble and removed in Jazzy
- Implement conditional compilation for jazzy
- Add packages that have not been released for jazzy in the repos file
- Promote jazzy support for packages that autoware depends on

### Definition of done

The initial Jazzy support is sufficient to build and minimally run with PlanningSimulator.
But making distructive changes for autoware behavior with ROS 2 Humble is not permitted.

HansRobo, 2024-06-20:
Updating dependencies and removing deprecated code for jazzy is going in #6656
NOTE: the pull request is going to be divided into small ones due to its largeness.

f0reachARR, 2024-06-20:
### Identified problems to be fixed
- [ ] Missing dependency that caused build errors
- [ ] Replace / introduce conditional include for deprecated header files
- [ ] Adopt changes of `optional` in Boost library
- [ ] Replace deprecated Ogre API in RViz plugins

### Identified problems, but not fixable now
- [ ] QoS arguments of create_client / create_service: In jazzy, setting it to rmw_qos_* is deprecated, but humble does not support new method.
- [ ] Some required packages are not provided by binary distribution.
    - [navigation2](https://github.com/ros-navigation/navigation2) (nav2_costmap_2d, etc...). Related issue: https://github.com/ros-navigation/navigation2/issues/4428
    - [grid_map](https://github.com/ANYbotics/grid_map)
    - [pacmod3_msgs](https://github.com/astuff/pacmod3_msgs)
    - Open3D (python3-open3d package). Related issue: https://github.com/isl-org/Open3D/issues/6730

HansRobo, 2024-06-20:
`ANYbotics/grid_map` will support Jazzy on v2.2.0
https://github.com/ANYbotics/grid_map/milestone/1

I heard from one of the maintainers of `ANYbotics/grid_map` that they are waiting for Jazzy support for `navigation2`.
https://x.com/wep21_/status/1794735182035218765

f0reachARR, 2024-06-20:
I just made a PR #7600 for fixing dependency errors.

yukkysaito, 2024-06-20:
Thank you for the suggestion. However, according to the community guidelines, issues are meant for reporting bugs. I will close this issue, so please repost it in the GitHub discussion.
https://autowarefoundation.github.io/autoware-documentation/main/support/support-guidelines/

closed 2024-06-20 by yukkysaito

HansRobo, 2024-06-20:
This is moved into https://github.com/orgs/autowarefoundation/discussions/4893