autowarefoundation/autoware_universe #7652: Evaluate current Localization accuracy of current NDT scan matching against Localization dataset

mitsudome-r, 2024-06-24:


mitsudome-r, 2024-06-24:
There is no guarantee that RTK is always accurate so I appreciate for any other way of getting ground truth data.

meliketanrikulu, 2024-08-15:
I tested with data which is this [PR](https://github.com/autowarefoundation/autoware-documentation/pull/597) shared.
Here is the localization broken points in [here](https://drive.google.com/file/d/18y8qgt8neHZeekriH2pU8tV5p1wL3uMp/view?usp=sharing) on the google earth.
You can open this .kml file on google earth.
Also you can see broken points on pointcloud map in [here](https://drive.google.com/file/d/16HxTj1WXSrDksJGPdtszkFh5LmEajGCZ/view?usp=sharing)
You can open this file with Cloud Compare.
I will also share test video.

meliketanrikulu, 2024-08-20:
I made some parametric changes and also we collected new data because of time problem in last data. You can check out the some test results with new data.
1.) To start, I simply wanted to observe how autoware localization works.  [Test Video](https://www.youtube.com/watch?v=Bk4Oyk6FOg0)

I did this test with Default autoware. Since I don't have the vehicle velocity as a Twist source, I did this test using the TwistWithCovarianceStamped message from GNSS/INS. 
There are 2 tunnels on our route. As expected, localization was disrupted while passing through the tunnels. Apart from this, no visible disruption was observed in localization at first glance.

2.) The error of GNSS/INS in the tunnel increases as seen in the video above.However, the frequency of the Twist message does not decrease or the Twist message is not interrupted. For this reason, I wanted to check if using the speed coming from here causes an error in the tunnel. For this reason, I tested the localization in the tunnel by giving only NDT poses as input to EKF. In this test, no Twist message was given to EKF. [Test Video](https://www.youtube.com/watch?v=ajgedIwwuaM)

meliketanrikulu, 2024-08-28:
I added Show&Tell Discussion for explaining to tests --> https://github.com/orgs/autowarefoundation/discussions/5135

meliketanrikulu, 2024-10-02:
Added document to [Autoware Documentation](https://autowarefoundation.github.io/autoware-documentation/pr-611/tutorials/components_evaluation/localization_evaluation/urban-environment-evaluation/) . We can close this issue .

closed 2024-10-02 by meliketanrikulu