autowarefoundation/autoware_universe #7724: Tracking disabled tests due to lack of gpu capable machines

xmfcx, 2024-06-27:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Story behind can be tracked from:
- https://github.com/autowarefoundation/autoware.universe/issues/7668#issuecomment-2194415792

Basically the tests disabled here:
- ✅ work on machines with Nvidia GPU
- ❌ fail on machines without Nvidia GPU

When we upgrade our testing infrastructure to have CUDA capable GPUs, we can re-enable these tests.

cc. @knzo25 @mitsudome-r 

### Purpose

To make CI pass on no-gpu machines. (Especially github hosted runners)

### Possible approaches

- Commenting out the failing tests in the respective CMakeLists.txt file.
- Adding a reference to here

### Definition of done

- [ ] `build-and-test` passes on the main branch.

xmfcx, 2024-06-27:
@mitsudome-r if somehow we gain access to gpu capable machines either by:
- adding more self hosted local machines
- or utilizing https://github.com/philips-labs/terraform-aws-github-runner

We can use this PR as reference:
- https://github.com/autowarefoundation/autoware.universe/pull/7721

I will also create a documentation on how I set up the self hosted runners with custom init scripts.

knzo25, 2024-08-30:
@xmfcx 
Is there any updates on this issue?
We would like to have the cuda tests enabled (https://github.com/autowarefoundation/autoware.universe/blob/main/perception/autoware_lidar_centerpoint/CMakeLists.txt)

xmfcx, 2024-09-03:
@knzo25 please have a peek at:
- https://github.com/autowarefoundation/autoware.universe/pull/8498

Through this new system, we have access to `gpu1.small` machines with `15 GB, 4vCPU` and an unknown GPU, I didn't make any tests.

Full list: https://aws.amazon.com/codebuild/pricing/ (select us east n. virginia)

But this machine is pricier.

We only need GPU for tests, not for building with CUDA.

So one option would be to separate build and test in the workflow and test on this new machine.

But it requires someone to work on this task.

xmfcx, 2024-12-10:
- https://github.com/autowarefoundation/autoware.universe/pull/5174 is tracked too.

mojomex, 2025-07-22:
This issue now has an intermediate solution: 
* #10999

While the long-term solution is to have GPU runners (an absolute necessity as more and more safety-critical functionality is being moved to GPU!), this PR executes all tests on every machine, regardless of GPU capabilities, but skips the tests if they require a GPU and there is none.

This has the following benefits over the status quo (commented out unit tests):

* unit tests are always compiled, ensuring that they stay up to date, even if the packages' source code changes
* unit tests are executed locally (assuming you have an Nvidia GPU), ensuring that any changes are tested at least locally
* unit tests in non-GPU CI runners show up as `SKIPPED`, clearly indicating that some functionality could not be tested, but without failing CI
* once GPU runners become available, unit tests will be executed without any further manual changes

I will do some PRs to modules that have commented-out unit tests.