autowarefoundation/autoware_universe #7877: [Dense-Urban-ODD] Improve cruise planner slowing down capabilities to overtaking road users

ahmeddesokyebrahim, 2024-07-05:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Part of:
- #7485

#### Dense Urban ODD Characteristics
- **Low-Speed Driving:** Typical speeds around 15 km/h.
- **Complex Traffic Elements:** Includes dense traffic with various dynamic interactions.
- **Dynamic Agents:** Frequent encounters with moving pedestrians and cyclists.
- **Bidirectional Movement:** Vehicles operate in lanes supporting bidirectional flow.
- **Traffic Rule Non-compliance:** Other road users may exhibit behaviors such as:
  - Driving in the wrong direction.
  - Cutting in unexpectedly from off-road areas.
  - Jaywalking pedestrians.
  - Improper / sudden overtaking by other road users.
  
In the context of Dense Urban ODD, we need to fine-tune the existing parameters of the `obstacle_cruise_planner` or potentially develop new features to improve the slowing down of ego vehicle to overtaking road user maintaining both passengers safety and comfortability.

### Purpose

#### Overtaking npc to ego lane then slowing down with different speeds
The following image describes the scenario where ... 
- an npc is driving in adjacent lane to ego. 
- then it passes the ego vehicle and overtakes the ego lane. 
- after overtaking, it starts to slow down.
[image]

The purpose of improvement in the following cases is to mitigate any unnecessary declaration (slowing down) value by ego vehicle while keeping the distance with overtaking npc.

##### NPC slowing down from 20 to 15 km/h
- https://youtu.be/qXYmOgy0s64

##### NPC slowing down from 20 to 12.5 km/h
- https://youtu.be/pMlVaiIYdDY

##### NPC slowing down from 20 to 10 km/h
- https://youtu.be/PhIGpvkZLWk

#### Bicycle driving through ego vehicle trajectory
The followingi mage describes the scenario where
- a bicycle is on the ego lane side (right or left)
- while ego vehicle is approaching, the bicylce start to move in a way overlapping with ego trajectory.
[image]

The purpose of improvement in the following cases is to slow-down for overtaking bicycle until enough space exists to overtake

##### Bicycle moving from right side of ego lane
- https://youtu.be/hQoJcC_f8n0

##### Bicycle moving from left side of ego lane
- https://youtu.be/xThCEhh0LbA

### Possible approaches

The `obstacle_cruise_planner` should be capable of handling the situations mentioned above. 
- The primary goal will be to tune the parameters of the package to achieve the success of above scenarios.
- If tuning the parameters is not enough for having successful scenarios, the  `obstacle_cruise_planner` implementation should be improved.

### Definition of done

- [ ] Create a proposal for changing parameters.
- [ ] If the package is inadequate to handle scenarios, propose improvements.
- [ ] Ensure all scenarios for Dense Urban ODD pass.

mitsudome-r, 2024-07-16:
This is the full length video of a scenario which doesn't abort when the failure condition is met. https://www.youtube.com/watch?v=c0ql4gGCVvo

ahmeddesokyebrahim, 2024-09-30:
Closing this issue as completed after merging the relevant [PR](https://github.com/autowarefoundation/autoware_launch/pull/1166).

closed 2024-09-30 by ahmeddesokyebrahim