autowarefoundation/autoware_universe #7916: Running source/install/setup.bash complains that it cannot find "local_setup.bash" for te

mitsudome-r, 2024-07-09:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Some packages that depends on TensorRT outputs error message as the following when I run `source/install/setup.bash` after building Autoware on machine without CUDA. 

Users can ignore it and run Autoware without an issue, but it would be better if we can make it so that it doesn't output any error for less confusion.

```
not found: "/home/mitsudome-r/autoware/install/tensorrt_common/share/tensorrt_common/local_setup.bash"
not found: "/home/mitsudome-r/autoware/install/tensorrt_classifier/share/tensorrt_classifier/local_setup.bash"
```

### Expected behavior

`source/install/setup.bash` finished without an error.

### Actual behavior

`source/install/setup.bash` outputs an error as described in the Description.

### Steps to reproduce

1. Install Autoware without installing CUDA
2. run `install/setup.bash`

### Versions

- OS: Ubuntu 22.04
- ROS 2: Humble
- Autoware: Main branch on July 9, 2024

### Possible causes

The packages don't call ament_package() nor ament_auto_package() in their CMakeLists.txt files.

Desperadozcx, 2024-09-23:
I also encountered this problem. Could you please tell me how you solved it?

AnzhiZhang, 2025-07-14:
Same issue, can ignore it and run Autoware without an issue.

```bash
root@SORA-WS3:/workspace# source install/setup.bash
not found: "/workspace/install/autoware_tensorrt_plugins/share/autoware_tensorrt_plugins/local_setup.bash"
```

However, a different thing is that I have installed it with CUDA.