autowarefoundation/autoware_universe #7923: [Dense-Urban-ODD] Adjust path planning and trajectory to follow direction of traffic flow in bi-directional lanes

lukewarmtemp, 2024-07-09:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Part of:
- https://github.com/autowarefoundation/autoware.universe/issues/7485#event-13148036468

### Characteristics:
- Vehicle is in a bi-directional lane
- Vehicle is expected to stay within the left or right lane (depending on the follow of traffic)
- Currently, the vehicle moves to the center of the lane

### Purpose

We need to fine-tune the existing parameters of `behaviour_path_planner` or potentially develop new features to allow the detection of bi-directional lanes and flow of traffic. This, in turn, will allow the vehicle to stay within either the left or right lane (depending on the flow of traffic) instead of moving to the center of the lane.

### Possible approaches

- The `behaviour_path_planner` module defines a path and drivable area the vehicle can move depending on the traffic situation. Investigation within this module may be a good starting place.

### Definition of done

- [ ] The vehicle to stay within the left or right lane in a bi-directional lane
- [ ] The vehicle detects the correct flow of traffic and follows the correct lane
- [ ] The vehicle to slow down if there are obstacles within the same lane, staying within the lane and not overtaking if there are obstacles in the other lane

mitsudome-r, 2024-08-19:
@lukewarmtemp Do you have any updates on this issue, or do you need someone's help?

lukewarmtemp, 2024-08-20:
> Do you have any updates on this issue, or do you need someone's help?

I'm sorry, I think some assistance would be helpful and appreciated. I've started trying to modify `behaviour_path_planner` but my scenario simulation doesn't seem to work anymore.

brkay54, 2024-08-28:
Hi, @lukewarmtemp is there any PR to fix this problem could you share?

beyzanurkaya, 2024-08-28:
Hi @lukewarmtemp, I can work on this task if you don't have time to focus on it.

closed 2024-09-30 by beyzanurkaya