autowarefoundation/autoware_universe #7955: Planned route is not connected and it makes the ego vehicle stuck.

Kim-mins, 2024-07-11:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi team,

I'm currently running Autoware with Carla using the bridge [here](https://github.com/autowarefoundation/autoware.universe/pull/6859#issuecomment-2196138040), and I found a situation that, the planned route is not connected and it makes the ego vehicle stuck.

Here are two videos for better descriptions: [[video 1](https://drive.google.com/file/d/1kQZo2UD5w4WVWXrFHnErT4u3yuxnSVpI/view?usp=sharing)] [[video 2](https://drive.google.com/file/d/1vVx1sz2eLnIPqlP1z0I-LxvlRWIUkmTB/view?usp=sharing)]

In video 1, which shows rviz while driving, the routing was already done, and the car starts to drive, but it stops since the drivable area is not connected. 
In video 2, which shows the whole route, I followed the route from the goal to the starting point, but they are not connected. (The route is quite complex, but this is because my current map does not support lane change.)

### Expected behavior

I hope the ego vehicle to drive well to the goal point.

### Actual behavior

But the car stops on the road, far from the destination.

### Steps to reproduce

Here's the ros2bag and map file: [[ros2bag](https://drive.google.com/drive/folders/14J2LAP7a7fRTMODn0jX0JhkhEx9RVbQq?usp=sharing)] [[map](https://github.com/hatem-darweesh/op_agent/blob/ros2/autoware-contents/maps/vector_maps/lanelet2/Town05.osm)]

### Versions

- autoware.universe: [7877192cb5700c1db15109dcd959785fc381951d](https://github.com/autowarefoundation/autoware.universe/tree/7877192cb5700c1db15109dcd959785fc381951d)
- Carla: 0.9.15

### Possible causes

I'm currently setting the goal point with [a routing api](https://autowarefoundation.github.io/autoware-documentation/main/design/autoware-interfaces/ad-api/list/api/routing/set_route_points/).
So I think this could be the cause of the bad routing result.

kyoichi-sugahara, 2024-07-17:
@Kim-mins 

Thank you for your reports. As one of the members developing the planning module, I greatly appreciate your bug reports.
Before we verify the problem on our end, I have one request. To determine if this is a Carla-specific issue, would it be possible for you to check if the same problem occurs in the simple planning simulator?
It would also be helpful if you could provide the autoware's version you're using.
While it's not a must, if you could also attach information about the start pose and goal pose by echoing the topics or through other means, it would save us time when reproducing the issue on our environment, which would be greatly appreciated.

Kim-mins, 2024-07-20:
Thank you for the response @kyoichi-sugahara!

> To determine if this is a Carla-specific issue, would it be possible for you to check if the same problem occurs in the simple planning simulator?

Actually I tried to, but I could not reproduce the issue with the planning simulator and even with Carla.
I provided the same starting/goal point to Autoware, but the result was ok.

> While it's not a must, if you could also attach information about the start pose and goal pose by echoing the topics or through other means, it would save us time when reproducing the issue on our environment, which would be greatly appreciated.

Sorry for providing too little information.
Here's the map I've used: [[map](https://github.com/hatem-darweesh/op_agent/blob/ros2/autoware-contents/maps/vector_maps/lanelet2/Town05.osm)] and I could retrieve the initial/goal point from the `launch.log` file as below:
```
# Initial point
x: 28.665252, 
y: 177.489838, 
z: 0.030679
qx: -0.000187, 
qy: 0.000211, 
qz: -0.716582, 
qw: 0.697503

# Goal point
x: -81.393974, 
y: -6.247448, 
z: 0.300000
qx: 0.000000, 
qy: 0.000000, 
qz: 0.001217, 
qw: 0.999999
```

Seems it is not reproducible on my side.. Could you please check if the same issue occurs with the map and initial/goal point?

Thank you!

kyoichi-sugahara, 2024-08-01:
@Kim-mins 

If the problem cannot be reproduced in your environment, I would like to lower the priority of this issue. 

If it becomes clear that the phenomenon can be reproduced in a specific version, I will investigate it myself. Until then, I'd like to proceed as follows: either you encounter a similar problem and share the reproduction conditions, or if someone encounter this phenomenon using a map, goal, and pose provided by you, I will investigate whether there's a cause on the planning side.

Until then, would it be alright to close this issue for now?

Kim-mins, 2024-08-08:
Thank you for the response @kyoichi-sugahara!

I agree to you. I cannot reproduce it now, and this error seems occur infrequently.
I guess this is from [a routing api](https://autowarefoundation.github.io/autoware-documentation/main/design/autoware-interfaces/ad-api/list/api/routing/set_route_points/), but I have no clue for now.

So I think we can close this issue now, and I'll let you know if I could encounter the similar situation.

Thank you!

closed 2024-08-08 by Kim-mins