autowarefoundation/autoware_universe #8032: The ego vehicle does not change lane and stops at a weird location.

Kim-mins, 2024-07-15:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi team,

I'm currently running Autoware with Carla using [bridge implementation](https://github.com/autowarefoundation/autoware.universe/pull/6859#issuecomment-2196138040), and I found a situation that the ego vehicle does not change lane and stops at a weird location.

Here's a rviz video of the situation: [[rviz](https://drive.google.com/file/d/1uSKjlTSt6e6rbnkXQTYDSSYA6WWCramr/view?usp=sharing)]

As shown in the video, the goal seems to require lane changes. However, the planned trajectory is not generated for lane change, and it drives forward, but it eventually stops at a weird location (seems the car stops at the location horizontally same with the goal point).

### Expected behavior

I hope the ego vehicle could drive to the goal point.

### Actual behavior

But it could not, and it eventually stops at a weird point.

### Steps to reproduce

Here's a ros2bag file for the reproduction: [[ros2bag](https://drive.google.com/drive/folders/1IAkF74d3Ji3WJb5e2Krv9oYaOi2xTubQ?usp=sharing)]

### Versions

- autoware.universe: [7877192cb5700c1db15109dcd959785fc381951d](https://github.com/autowarefoundation/autoware.universe/tree/7877192cb5700c1db15109dcd959785fc381951d)
- Carla: 0.9.15

maxime-clem, 2024-07-15:
The map probably does not allow lane change.
Can you share your lanelet map and I will confirm if this is the issue.

Kim-mins, 2024-07-15:
Thank you for the response @maxime-clem!

Here's the map file: [[map](https://drive.google.com/file/d/11BHovDC-ir2SjmcI-QoVnG531EBqSFRc/view?usp=sharing)] I forgot to upload it.. sorry for that.
Actually, I manually modified only those(of the video above) lanes of the map(from [here](https://github.com/hatem-darweesh/op_agent/blob/ros2/autoware-contents/maps/vector_maps/lanelet2/Town04.osm)) to support lane change, and I checked that the lane change could be successful with the modified map. 
Here's the video of successful lane changing: [[video](https://drive.google.com/file/d/15SQiW0YJvvEFB7Sp_C3IBM_a_q924uuy/view?usp=sharing)]

Thanks!

takayuki5168, 2024-07-17:
@Kim-mins 
Can you give us the video with the updated map where the routing failed?

Kim-mins, 2024-07-17:
Hi @takayuki5168!

Sorry I could not get your point.. Could you please clarify more for me?

In my experience, for the route requires lane change, routing failure occurs when the map is not updated(=not modified for lane change), and when I updated the map, as the map supports lane change, the routing was successful.

In short, in my experience, there could be routing failure with original map or routing success with updated map. So I could not get the video for routing failure on updated map.. sorry.