autowarefoundation/autoware_universe #8033: Autoware keeps acceleration less than 0 and the ego vehicle does not drive at all from the starting point.

Kim-mins, 2024-07-15:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi team,

I'm currently running Autoware with Carla using [bridge implementation](https://github.com/autowarefoundation/autoware.universe/pull/6859#issuecomment-2196138040), and I found a situation that Autoware keeps acceleration less than 0 and the ego vehicle does not drive at all from the starting point.

Here's the rviz video of the situation: [[video](https://drive.google.com/file/d/1NpCeUIaOOOr9vTVTIxGWleLMMyGYuDpi/view?usp=sharing)] and corresponding `launch.log` file: [[launch.log](https://drive.google.com/file/d/1jQ3YVHXiK3QGfAmySIGKtoykGpshXys2/view?usp=sharing)]

When I checked the control command by subscribing the topic `/control/command/control_cmd` and plotted `acceleration` and `speed` as below, it showed that the acceleration is below 0 and  the speed does not increase (the values oscillates though..).
[image]

### Expected behavior

I hope the ego vehicle to drive well to the goal point.

### Actual behavior

But the ego vehicle does not start to drive.

### Steps to reproduce

Here's the ros2bag file for the reproduction: [[ros2bag](https://drive.google.com/drive/folders/1AMlawy_GO8FMCRlOC4Vfk7EELuMcGn8G?usp=sharing)]

### Versions

- autoware.universe: [7877192cb5700c1db15109dcd959785fc381951d](https://github.com/autowarefoundation/autoware.universe/tree/7877192cb5700c1db15109dcd959785fc381951d)
- Carla: 0.9.15

### Additional context

I thought this issue is not duplicated with https://github.com/autowarefoundation/autoware.universe/issues/8032, as no logs in `launch.log` says process died.

VRichardJP, 2024-07-17:
Your logs basically show 3 errors:

```
1720950077.5001881 [component_container_mt-64] [ERROR] [1720950077.499698771] [compressed_depth_image_transport]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
1720950077.5126553 [component_container_mt-6] [WARN] [1720950077.511458876] [sensing.lidar.concatenate_data]: transformed_raw_points[/sensing/lidar/left/pointcloud] is nullptr, skipping pointcloud publish.
1720950077.5150239 [component_container_mt-6] [WARN] [1720950077.511521889] [sensing.lidar.concatenate_data]: transformed_raw_points[/sensing/lidar/right/pointcloud] is nullptr, skipping pointcloud publish.
```

Concatenated lidar data is required by the perception pipeline, so the concatenation issue could be blocking in your case. 

Did you check Autoware diagnostics with RQT for other potential issues?

Kim-mins, 2024-07-17:
Thank you for the response @VRichardJP!

Actually, I did not check the whole perception pipeline, but I checked that the car drives well in most of the cases even with the messages above, and it is not that frequent to stuck at the starting point.

To my knowledge, those error messages are from the [bridge implementation](https://github.com/autowarefoundation/autoware.universe/pull/6859#issuecomment-2196138040) as the bridge is not handling the topics `/sensing/lidar/left/pointcloud` and `/sensing/lidar/right/pointcloud`.
I'll check if those errors are relevant to this issue soon.

Thanks!

Kim-mins, 2024-07-17:
Hi @VRichardJP!

When running simulation on Carla with the bridge, there's the only LiDAR for the ego vehicle due to performance issue, so it seems we need not to process those topics for now. (Thank you for checking @maxime-clem!)