autowarefoundation/autoware_universe #8143: Avoidance by lane change not working

Joonki0111, 2024-07-22:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I'm currently using Autoware.Universe release/v1.0 with Ubuntu22.04 + Humble.
As I was testing the avoidance module, somehow I cannot execute the avoidance_by_lane_change_module. 
If I launch the planning simulator and create a dummy car in front of ego vehicle, it just follows the dummy car by obstacle_cruise_planner.
[image]

It looks like it is never approved? by the planner_manager. I have managed to change some parameters like 
### execute_object_longitudinal_margin = 0.0
### execute_num = 0
but i'm not sure exactly where is the problem.

### Expected behavior

Avoid target object by changing lane

### Actual behavior

Cruise along with the target object

### Steps to reproduce

Planning simulator with dummy car

### Versions

Ubuntu22.04 + Humble

### Possible causes

As far as I know, 
`
bool AvoidanceByLaneChangeInterface::isExecutionRequested() const
{
  return module_type_->specialRequiredCheck() && module_type_->isLaneChangeRequired();
}
`
those two returns in behavior_path_avoidance_by_lane_change_module package are both false and 
specialRequiredCheck() is false by the code below.
`  if (!status_.is_safe) {
    return false;
  }`

Joonki0111, 2024-07-22:
Additionally, the lanelet is enabled to change lane.

felixf4xu, 2024-08-01:
the code changed a lot in recent branches, like in `main` branch, did you tried the latest version?

Joonki0111, 2024-08-01:
@felixf4xu Yes, I have tried the release/2024.07 branch. The release/2024.07 branch was able to avoid the "static" obstacle by avoidance module. But avoiding "dynamic" obstacle by changing lane was never executed in any of scenario. I was expecting to avoid dynamic object by changing lane but I found one documentation.
[image]
maybe it is just not possible right now.

felixf4xu, 2024-08-01:
I'm having issues on the same module, check out my issue https://github.com/autowarefoundation/autoware.universe/issues/8274

mehmetdogru, 2024-08-01:
@Joonki0111 

Thank you for opening up the issue.

Could you please share your HD-Map where we can reproduce the issue?

Joonki0111, 2024-08-01:
@mehmetdogru Sorry, I already found the answer to my issue. Avoiding dynamic object on same lane using avoidance by lane change module is under development which means it is not possible currently.

mehmetdogru, 2024-08-01:
@Joonki0111 

If your target object is moving, that is correct. If so could you please close the issue?

Joonki0111, 2024-08-01:
@felixf4xu  I've checked your issue and it seems like a bug or something(I might be wrong). Actually, your issue and mine has difference in terms of avoiding static or dynamic. I'm not sure with the avoidance by lane change module, it might be still in development.

closed 2024-08-01 by Joonki0111