autowarefoundation/autoware_universe #8219: lidar_centerpoint does not compile with CUDA 12.5

PPeltzerFka, 2024-07-26:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Building Universe fails for me when following the [Source installation](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/source-installation/) guide, with running the `./setup-dev-env.sh` on Ubuntu 22, installing the latest Cuda 12.5 (released in May). When trying to build Autoware Universe, the compilation fails for lidar_centerpoint, pointing to missing declaration for the `__syncthreads()` function and more.

### Expected behavior

Following the tutorial, compilation is expected to be successful.

### Actual behavior

The compilation fails for lidar_centerpoint. Unfortunately, I only have a part of the error message right now as I dont want to reinstall CUDA 12.5 again to track the bug. This one was originating fro test_preprocess_kernel.cpp :

```
/usr/local/cuda/include/cub/util_ptx.cuh:271:5: error: ‘__syncthreads’ was not declared in this scope
  271 |     __syncthreads();
```

### Steps to reproduce

1. Run ./setup-dev-env.sh
2. Validate you're running CUDA 12.5 (`nvcc --version`)
3. Build autoware universe, or only up to lidar_centerpoint: `colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release --packages-up-to autoware_lidar_centerpoint`

### Versions

- OS: Ubuntu 22
- ROS 2: Humble
- Autoware: Initially tried with 2024.06 tag, but issue appears to be still there in latest Universe main commit

### Possible causes

Google spits out the following threads related to this:

- [Unable to include thrust/host_vector.h and others with CUDA 12.5](https://stackoverflow.com/questions/78784236/unable-to-include-thrust-host-vector-h-and-others-with-cuda-12-5)
- [When upgrade from CUDA12.4 to 12.5 the compilation became broken](https://forums.developer.nvidia.com/t/when-upgrade-from-cuda12-4-to-12-5-the-compilation-became-broken/295814)

There appears to be an API change in the thrust library released in CUDA 12.5, no longer allowing the inclusion of certain (?) headers in the .cpp/.hpp files instead of .cu files (judging from the linked topics).

lidar_centerpoint includes thrust libraries in a couple of files:

test_preprocess_kernel.cpp
```
#include <thrust/host_vector.h>
#include <thrust/reduce.h>
```

circle_nms_kernel.hpp
```
#include <thrust/device_vector.h>
```

postprocess_kernel.hpp
```
#include <thrust/device_vector.h>
```

test_preprocess_kernel.cpp
```
#include <thrust/host_vector.h>
#include <thrust/reduce.h>
```

Note that the lidar_transfusion package has the same include pattern, so the issue is probably present there as well.

### Additional context

Removing CUDA 12.5 and installing CUDA 12.4 fixes the issue.

amadeuszsz, 2024-07-29:
Hi @PPeltzerFka, thank you for your detailed report.

Currently Autoware is developed with [CUDA 12.3](https://github.com/autowarefoundation/autoware/blob/ede026d9dd799b5372fd20cb4d916950e1a49cb9/amd64.env#L4).  I can reproduce your issue and seems CUDA 12.5 came with API breaking changes. For the time being, could you use CUDA 12.3 / 12.4?  Also here is another [link](https://github.com/NVIDIA/cccl/issues/1373) which cover this issue.

amadeuszsz, 2024-07-31:
Hi @PPeltzerFka 
We made a fix PRs. I will appreciate if you could let us know later you can build and run Autoware successfully on your machine 🙇🏻‍♂️

closed 2024-08-03 by amadeuszsz