autowarefoundation/autoware_universe #8274: `avoidance by lane change` module issues

felixf4xu, 2024-07-31:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi,

I'm testing the avoidance by lance change module but I have several issues.

[image]

My test setup:

I **disabled** all other "behavior path modules" and only left "avoidance by lane change" as true like this (in launcher/autoware_launch/autoware_launch/config/planning/preset/default_preset.yaml):
```yaml
  - arg:
      name: launch_avoidance_by_lane_change_module
      default: "true"
```

I set a goal on the left side lane of the starting initial position, then set an "2D Dummy car" in front of the ego vehicle. The result is show as in the screenshot. In rviz2, I checked the `PathChangeCandidate_AvoidanceByLC` and it's displayed.

### Expected behavior

Issue 1: 

the "virtual wall" of `avoidance_by_lane_change` is close to the goal, not the obstacle. It it correct? If I move the goal to a further position, the virtual wall will move along with the goal. Should it be close to the obstacle?

Issue 2:

Should I disable `obstacle_stop` module? I see a virtual wall of it.

issue 3:

In this setup, If I click "auto" to start the planning simulation, the ego vehicle will start and drive in its path and then stops in front of the dummy car, it actually does not avoid or change the lane.

### Actual behavior

The ego car should change the lane and avoid the obstacle.

### Steps to reproduce

check the screenshot I provided.

### Versions

Ubuntu 22
ros2 Humble
Autoware, main branch.

### Possible causes

Different module settings? Planner Priorities? Enable other modules?

mehmetdogru, 2024-08-01:
@felixf4xu Thanks for creating the issue.

_Issue 1:_ Surely the virtual wall is not created in right place. Virtual wall for `avoidance_by_lane_change` should be before the object.

_Issue 2:_ No need to disable `obstacle_cruise_planner` to test your case

Could you share your HD-Map and create a scenario if possible (using https://scenario.ci.tier4.jp/scenario_editor) so that this issue can be reproducible easily by the developers?

felixf4xu, 2024-08-02:
It seems that by PR https://github.com/autowarefoundation/autoware.universe/pull/7436, `waitApproval()`/`clearWaitingApproval()` pair is deleted from the base class of `LaneChangeInterface::plan()`, then it's lane change module's (in my case, it's `avoidance_by_lane_change` module) responsibility to call `clearWaitingApproval()`.

But in the latest code, `clearWaitingApproval()` is never called.

zulfaqar-azmi-t4, 2024-08-06:
The current behavior is as intended.
The current virtual wall position is positioned at the terminal start point. 

The Avoidance by lane change module is not well refined enough, therefore it doesn't insert the stop point before obstacles.

closed 2024-08-08 by felixf4xu