autowarefoundation/autoware_universe #8389: `traffic_light_roi_visualizer` does not handle signal with arrow indicator (in Japan)

YoshiRi, 2024-08-07:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

`traffic_light_roi_visualizer` does not show entire detection results when following traffic lights.

[image]
[image]

We want visualizer to show every traffic light detection results.
That would be something like:

```
●: 100%
← : 100%
↑: 100%
```

### Purpose

We want to distinguish detection error and visualization error.

### Possible approaches

First of all, all of the detection results should be visualized.
The easiest solution is to concatenate visualization text with vertically or horizontally.

```
●: 100%
← : 100%
↑: 100%
```

```
●, ←, ↑
100, 100, 100
```

### Definition of done

When we confirmed all of the detection results should be visualized.

Currently TIER IV has the sample data. [TIER IV INTERNAL DATA LINK](https://star4.slack.com/archives/C076ZGRC15X/p1722936498548229?thread_ts=1722932121.436609&cid=C076ZGRC15X), if there are someone interested in this issue, we will annonimise and share this rosbag.

kminoda, 2024-08-08:
@kaancolak 
Hi, would you address this issue? I think this is due to your PR: https://github.com/autowarefoundation/autoware.universe/pull/6985. Since this may cause misunderstanding of the Autoware's performance, we would like to fix it as soon as possible. What we would like to see is that all the possible patterns as a traffic light classification output is properly checked.
(One of our members within TIER IV approved your PR, but should've checked from our end, sorry for that)

Thank you for your contribution as always :pray:

kaancolak, 2024-08-08:
Hi @kminoda -san, this doesn't seem to be related to me. Maybe @KhalilSelyan might be able to help

kminoda, 2024-08-08:
Sorry for that :pray: Fixed

ktro2828, 2024-08-09:
Hi I've created a PR to fix this problem. Could you review that? https://github.com/autowarefoundation/autoware.universe/pull/8428

ktro2828, 2024-08-16:
https://github.com/autowarefoundation/autoware.universe/pull/8428 has been merged

YoshiRi, 2024-08-27:
I will close this issue with https://github.com/autowarefoundation/autoware.universe/pull/8428 has been merged

closed 2024-08-27 by YoshiRi