autowarefoundation/autoware_universe #8513: external request lane change not working properly

Joonki0111, 2024-08-18:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

If moving to side lanes by [behavior_path_external_request_lane_change_module](https://github.com/autowarefoundation/autoware.universe/tree/release/v1.0/planning/behavior_path_external_request_lane_change_module), the module doesn't work as expected.

1. Path planned with both left and right as expected
[image]

2. Ego triggered to change lane 
[image]

3. In middle of changing lane 
[image]

4. Successfully changed lane with no available path candidate 
[image]
    But in debug info, external_request_lane_change_left module can be found at candidate module
[image]

At first I thought external request lane change in a row was impossible but found out that one side to another was possible.
In other words, lane change direction can't be shifted. 

### Expected behavior

change lane freely with [behavior_path_external_request_lane_change_module](https://github.com/autowarefoundation/autoware.universe/tree/release/v1.0/planning/behavior_path_external_request_lane_change_module)

### Actual behavior

Can't change direction of lane change

### Steps to reproduce

Any map with planning simulator

### Versions

Ubuntu22.04 Humble 
Autoware release/v1.0

zulfaqar-azmi-t4, 2024-08-19:
@Joonki0111 

Thank you so much for the report.
Is the goal in the middle lane?

The unlike normal LC, external request LC typically searches for non-preferred lane to use as target. If the goal is in the middle lane, it will become preferred lane, hence external request LC will not to be able to find its  target lane.

Joonki0111, 2024-08-19:
@zulfaqar-azmi-t4 Thanks for the reply. I found the answer couple of hours ago which is same as your reply. I'm now currently working on modifying the /planning/mission_planning/route topic when external request LC is processed to change lane in any situation. Glad to hear it's not a bug.

closed 2024-08-21 by Joonki0111