autowarefoundation/autoware_universe #8530: Whether there is a bug in the frame_id of the perception module output objects

cyn-liu, 2024-08-19:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I set perception_mode=`lidar` and use `lidar_centerpoint` for object detection, but I find that my detection box does not show up on the lidar target.
I debug and found that the `frame_id` of  the output topic `/perception/object_recognition/detection/centerpoint/objects`  is `velodyne_top`, I think this is unreasonable, it should be `map` or `base_link`, which is configured by param `world_fram_id` in the file `centerpoint.param.yaml`.

**I have correctly configured the extrinsic parameters from `velodyne_top` to `base_link`, and NDT localization  is successful.**

### Expected behavior

The output topic `/perception/object_recognition/objects` of the perception module is drawn on the point cloud object.

### Actual behavior

The output topic `/perception/object_recognition/objects` of the perception module is not drawn on the point cloud object.

`/perception/object_recognition/detection/centerpoint/objects`
[image]

`/perception/object_recognition/detection/clustering/objects`
[image]

`/perception/object_recognition/detection/temporary_merged_objects`
[image]

`/perception/object_recognition/detection/objects_before_filter`
[image]

`/perception/object_recognition/detection/objects`
[image]

`/perception/object_recognition/objects`
[image]

### Steps to reproduce

1. modify input/pointcloud topic
2. build Autoware
3. launch Autoware

### Versions

OS: amd
ROS2: humle
Autoware: The latest autoware.universe pulled from the autoware repository

cyn-liu, 2024-08-21:
lidar2ego external parameters and localization is correct:
[image]

The display result of topic `/perception/object_recognition/objects` in rviz2
[image]
[image]

vividf, 2024-08-28:
@cyn-liu 
Thanks for your report!
I was able to reproduce your result and will fix the bug soon.

vividf, 2024-08-29:
@cyn-liu The issue can be solved with PR https://github.com/autowarefoundation/autoware.universe/pull/8674.
Thanks for your report!