autowarefoundation/autoware_universe #8589: After performance improvements of autoware_pointcloud_preprocessor voxel_based_compare_map_filter did not work

sebekx, 2024-08-22:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

In latest perf improvements of `autoware_pointcloud_preprocessor` [in this PR](https://github.com/autowarefoundation/autoware.universe/pull/8174) changed the way how tf are retrieved in `filter` class, which is used i.e. in  `voxel_based_compare_map_filter`. New improvements assumes that only static tf's are used, however in  `voxel_based_compare_map_filter` there is transformation from `base_link` to `map`.

### Expected behavior

`voxel_based_compare_map_filter` filter pointcloud based on poincloud map. correctly transform input pointcloud to `map` frame and filter matching points and transform back to `base_link` 

### Actual behavior

`voxel_based_compare_map_filter` after start correctly filter pointcloud but after vehicle moves old tf is used which cause filter to not match input pointcloud points with map points

### Steps to reproduce

1 Start autoware
2. Enable  /perception/object_recognition/detection/pointcloud_map_filtered/pointcloud topic in rviz
3. Observe filtered points after vehicle moves

### Possible causes

performance improvements of autoware_pointcloud_preprocessor [in this PR](https://github.com/autowarefoundation/autoware.universe/pull/8174) which retrieves tf only once which in case of `base_link` <-> `map` transform is incorrect

badai-nguyen, 2024-08-28:
@amadeuszsz Could you check this issue as it seems related to [your PR](https://github.com/autowarefoundation/autoware.universe/pull/8174)

amadeuszsz, 2024-09-09:
PR has been merged, closing.

closed 2024-09-09 by amadeuszsz