autowarefoundation/autoware_universe #8650: Missing drivable area when two lanelets merged into one

felixf4xu, 2024-08-28:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I have a map where 2 lanes will merge into one, like this:
[image]
and it can be simplified as this:
[image]

### Expected behavior

on dashed lines, lane change should be possible, all area should be drivable.

### Actual behavior

[image]
There is some triangle area (circle in red), which is not drivable

### Steps to reproduce

the lanelet map is uploaded
[lanelet2_map.zip](https://github.com/user-attachments/files/16775942/lanelet2_map.zip)

### Versions

- Ubuntu 22
- Humble
- autoware main branch

### Possible causes

in the overlapped area, it's difficult to tell which lanelets are neighbors

felixf4xu, 2024-08-28:
First of all, I would like to check if the map (lanelet) is drawn correctly, I mean, maybe there are some other ways to draw the lanelets so that autoware can successfully understand.

mehmetdogru, 2024-08-29:
Thank you @felixf4xu for creating the issue. 

The map seems right to me. 

For path planner to expand drivable area so that neighbor lanes as well would be covered, they should share a linestring at the moment. Since in your drawings lanelet 2 and 3 does not share a linesting drivable area is not expanded. I am not sure how easy to implement what you want.

felixf4xu, 2024-08-30:
That missing triangle area actually does not matter much when driving at 4-2-1 or 5-3-1, but it matters when checking if lane change is possible from 4 -> 5 or from 5 -> 4. 

[image]

maxime-clem, 2024-08-30:
The map is fine but the current planning modules do not handle this case well.
To do the lane change with a merging lane, it is better to draw lanelet 2 as a triangle such that it does not have a successor but has an adjacent lanelet (required to do lane change).
[image]

I tested it and now we can do a lane change.
[image]

I could not find anything about merging lanes in the map creation requirements (https://docs.pilot.auto/en/reference-design/common/map-requirements/overview) so we should probably keep this issue open and either:
- Add a requirement for merging lanes in the documentation.
- Properly handle merging lanes represented with 2 lanelets preceding one lanelet.

felixf4xu, 2024-10-31:
> I tested it and now we can do a lane change

Hi @maxime-clem , did you test it with some new code added? is the code already merged into some branch of autoware.universe?

maxime-clem, 2024-10-31:
> did you test it with some new code added?

No new code. I only modified the map file as I described.