autowarefoundation/autoware_universe #8739: [Dense-Urban-ODD] Improve avoidance for moving NPCs in same and opposite direction of ego driving

ahmeddesokyebrahim, 2024-09-03:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Part of:
- #7485

### Dense Urban ODD Characteristics
- **Low-Speed Driving:** Typical speeds around 15 km/h.
- **Complex Traffic Elements:** Includes dense traffic with various dynamic interactions.
- **Dynamic Agents:** Frequent encounters with moving pedestrians and cyclists.
- **Bidirectional Movement:** Vehicles operate in lanes supporting bidirectional flow.
- **Traffic Rule Non-compliance:** Other road users may exhibit behaviors such as:
  - Driving in the wrong direction.
  - Cutting in unexpectedly from off-road areas.
  - Jaywalking pedestrians.
  - Improper overtaking by drivers.
  - Moving NPCs in side of the lane either in same or opposite direction of driving
  
In the context of Dense Urban ODD, we need to fine-tune the existing features of the `autoware_behavior_path_dynamic_obstacle_avoidance_module`  and potentially develop new features to better avoid moving NPCs in dense urban areas.

### Purpose

#### Moving NPC in the same direction of ego "UC-NTR-002-0001" "2nd and 3rd iterations"
The following image describes the scenario where
 
[image]

#### Moving NPC in opposite direction "UC-NTR-002-0003"
[image]

### Possible approaches

The `autoware_behavior_path_dynamic_obstacle_avoidance_module`  should be capable of handling the situations mentioned above.
- The primary goal will be to tune the parameters of the package to achieve the success of above scenarios.
- If tuning the parameters is not enough for having successful scenarios, the `autoware_behavior_path_dynamic_obstacle_avoidance_module` implementation should be improved.

### Definition of done

### Definition of done

- [ ] Create a proposal for changing parameters.
- [ ] If the package is inadequate to handle scenarios, propose improvements.
- [ ] Ensure all scenarios for Dense Urban ODD pass.

mitsudome-r, 2024-09-10:
@ahmeddesokyebrahim handles the static objects, and @beyzanurkaya will be working on the dynamic obstacle handling.

ahmeddesokyebrahim, 2024-09-10:
> @ahmeddesokyebrahim handles the static objects, and @beyzanurkaya will be working on the dynamic obstacle handling.

@mitsudome-r -san :
The static obstacle cases are handled in this [issue](https://github.com/autowarefoundation/autoware.universe/issues/8539) and [PR](https://github.com/autowarefoundation/autoware_launch/pull/1129)

closed 2024-11-26 by xmfcx