autowarefoundation/autoware_universe #8752: Statistics of latest Autoware's Whole Perception Pipeline Latency (Lidar only mode)

cyn-liu, 2024-09-04:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I've agreed with the maintainers that I can plan this task.

### Description

Counting the running time of the perception module(lidar-only mode) during real vehicle operation.

Environment: 
Device: DELL precison3581
OS: Ubuntu22.04
GPU: NVIDIA RTX A1000 6GB Laptop
CPU: intel CORE i7

### Purpose

In order to get a detailed of each node and the overall time consumption of the perception pipeline, and to facilitate developers in improving the performance of the perception module.

### Possible approaches

We will investigate a reasonable method to calculate the running time of the perception pipeline.
**@mitsudome-r  @xmfcx** 
Do you have any suggestions for statistical tools and methods?

### Definition of done

TODO

xmfcx, 2024-09-04:
### CARET

https://github.com/orgs/autowarefoundation/discussions/2822

### Pipeline timings 

- https://github.com/autowarefoundation/autoware.universe/issues/6032

- https://github.com/autowarefoundation/autoware.universe/issues/5540

cyn-liu, 2024-10-12:
We use the `Pipeline timings` statistical method recommended by xmfc above to perform latency time statistics on Autoware's perception pipeline(only lidar mode).

Autoware perception module publish the following `pipeline_latency_time` debug information, we subscribe to these topics and analyze numerical changes through `rqt` to evaluate the time consumption of the perception pipeline.
```
/perception/obstacle_segmentation/crop_box_filter/debug/pipeline_latency_ms
/perception/occupancy_grid_map/raw_pc_downsample_filter/debug/pipeline_latency_ms
/perception/occupancy_grid_map/obstacle_pc_downsample_filter/debug/pipeline_latency_ms
/perception/occupancy_grid_map/pointcloud_based_occupancy_grid_map/debug/pipeline_latency_ms
/perception/object_recognition/detection/clustering/voxel_grid_based_euclidean_cluster/debug/pipeline_latency_ms
/perception/object_recognition/detection/centerpoint/lidar_centerpoint/debug/pipeline_latency_ms
/perception/object_recognition/detection/obstacle_pointcloud_based_validator/debug/pipeline_latency_ms
/perception/object_recognition/detection/object_lanelet_filter/debug/pipeline_latency_ms
/perception/object_recognition/tracking/multi_object_tracker/debug/pipeline_latency_ms
```

Scenario1: Test road with few other traffic participants

https://github.com/user-attachments/assets/c1119fc7-e031-4c4d-8b2f-f5d9eba480a1

Observing the statistical results, we found that:

`/perception/object_recognition/tracking/multi_object_tracker/debug/pipeline_latency_ms`  is the most time-consuming, with an average time of approximately 200ms.

```
/perception/obstacle_segmentation/crop_box_filter/debug/pipeline_latency_ms
/perception/occupancy_grid_map/raw_pc_downsample_filter/debug/pipeline_latency_ms
/perception/occupancy_grid_map/obstacle_pc_downsample_filter/debug/pipeline_latency_ms
/perception/occupancy_grid_map/pointcloud_based_occupancy_grid_map/debug/pipeline_latency_ms
```
The average time for A is about 40ms.

```
```
The average time for A is about 60ms.

``
The average time for A is about 40ms.

cyn-liu, 2024-10-15:
Scenario 2: Urban road with many other traffic participants

https://github.com/user-attachments/assets/60bf8a1a-0ee4-47eb-af54-7df368daf768

We use box plots to analyze the following topics related to perception module:
```
/perception/obstacle_segmentation/crop_box_filter/debug/pipeline_latency_ms
/perception/occupancy_grid_map/raw_pc_downsample_filter/debug/pipeline_latency_ms
/perception/occupancy_grid_map/obstacle_pc_downsample_filter/debug/pipeline_latency_ms
/perception/occupancy_grid_map/pointcloud_based_occupancy_grid_map/debug/pipeline_latency_ms
/perception/object_recognition/detection/clustering/voxel_grid_based_euclidean_cluster/debug/pipeline_latency_ms
/perception/object_recognition/detection/centerpoint/lidar_centerpoint/debug/pipeline_latency_ms
/perception/object_recognition/detection/obstacle_pointcloud_based_validator/debug/pipeline_latency_ms
/perception/object_recognition/detection/object_lanelet_filter/debug/pipeline_latency_ms
/perception/object_recognition/tracking/multi_object_tracker/debug/pipeline_latency_ms
```
[image]
[image]
[image]

As shown in the above figure, the average time consumption of `/perception/obstacle_segmentation/crop_box_filter/debug/pipeline_latency_ms` and `/perception/occupancy_grid_map/raw_pc_downsample_filter/debug/pipeline_latency_ms` and `/perception/occupancy_grid_map/obstacle_pc_downsample_filter/debug/pipeline_latency_ms` and `/perception/occupancy_grid_map/pointcloud_based_occupancy_grid_map/debug/pipeline_latency_ms` and `/perception/object_recognition/detection/centerpoint/lidar_centerpoint/debug/pipeline_latency_ms` is about 50ms.

As shown in the above figure, the average time consumption of 
`/perception/object_recognition/detection/clustering/voxel_grid_based_euclidean_cluster/debug/pipeline_latency_ms` and `/perception/object_recognition/detection/obstacle_pointcloud_based_validator/debug/pipeline_latency_ms` and `/perception/object_recognition/detection/object_lanelet_filter/debug/pipeline_latency_ms` is about 70ms.

As shown in the above figure, the average time consumption of 
`/perception/object_recognition/tracking/multi_object_tracker/debug/pipeline_latency_ms`  is about 130ms.

cyn-liu, 2024-10-29:
We found that the `multi_object_tracker` node is very time-consuming, so we analyzed the various modules inside the node, a lot of debug information has been published within the node to help us analyze the time consumption.
1. `/perception/object_recognition/tracking/multi_object_tracker/debug/input_latency_ms`
Indicate the time delay from receiving data to starting data processing.

2. `/perception/object_recognition/tracking/multi_object_tracker/debug/processing_latency_ms`
Indicate the time delay from data processing to data publishing.

3. `/perception/object_recognition/tracking/multi_object_tracker/debug/processing_time_ms`
The time from the start of publishing to the completion of publishing + the time from the start of processing to the end of processing.
But I don't know why the value of this topic is equal to 0.

4. `/perception/object_recognition/tracking/multi_object_tracker/debug/meas_to_tracked_object_ms`
Indicate the time delay from receiving data to triggering the timer (before data publishing).

5. `/perception/object_recognition/tracking/multi_object_tracker/debug/pipeline_latency_ms`
Indicate the time delay from receiving data to publishing.

6. `/perception/object_recognition/tracking/multi_object_tracker/debug/cyclic_time_ms`
Indicate the time interval between two pubs.

[image]
[image]
[image]

From the above figure, it can be seen that `/perception/object_recognition/tracking/multi_object_tracker/debug/pipeline_latency_ms` takes approximately `125 ms`, and it can be roughly assumed that `pipeline_latency_ms` = `input_latency_ms` + `processing_latency_ms`, in which,  the time consumption of `processing_latency_ms` is relatively stable, about `75 ms`, while the time consumption of `input_latency_ms` fluctuates greatly, usually around `50 ms`.

cyn-liu, 2024-11-06:
After analysis, the reason for the long time consumption of node `multi_object_tracker` has been identified:
From the above analysis, it can be seen that `/perception/object_recognition/tracking/multi_object_tracker/debug/pipeline_latency_ms` takes approximately 125ms, which is mainly composed of `input_latency_ms` and `processing_latency_ms`.
1. For `input_latency_ms`, it refers to the time consumption of data preprocessing, calculated as follows: 
```cpp
double input_latency_ms = (now - last_input_stamp_).seconds() * 1e3;
```
`now` refers to the time after data preprocessing.
`last_input_stamp_` refers to the timestamp of the earliest received data in the data queue.

After analysis, it was found that the data preprocessing function is not time-consuming, so the reason why `input_latency_ms` takes a lot of time is that the data in the data queue is too old.

2. For `processing_latency_ms` , it refers to the time taken between starting the objects tracking algorithm and publishing the results, calculated as follows: 
```cpp
double processing_latency_ms = ((now - stamp_process_start_).seconds()) * 1e3;
```
`now` refers to the time after publishing the results.
`stamp_process_start_` refers to the time to start executing the tracking algorithm.

After analysis, it was found that the time consumption of each function in the tracking algorithm is very small. The main source of time consumption for `processing_latency_ms` is that the publishing thread is triggered by a timer, while the tracking algorithm thread is triggered by data. The two are not synchronized, resulting in a high time consumption.

cyn-liu, 2024-11-06:
After analysis, I believe that there is an error in the calculation method of `/perception/object_recognition/tracking/multi_object_tracker/debug/input_latency_ms`,  In its calculation formula, `last_input_stamp_` refers to  the timestamp of the earliest received data in the data queue. but the objects data received by this node is not the timestamp at the time of publication, which will render this calculation formula meaningless.

cyn-liu, 2024-11-13:
`pipeline_latency_ms` represents the time it takes for the entire pipeline from point cloud publishing to the completion of execution at the current node.  During the statistical process, it was found that node `multi_object_tracker`'s `/perception/object_recognition/tracking/multi_object_tracker/debug/pipeline_latency_ms` is very high and fluctuates greatly. After analysis, it was found that this is because the data received by node `multi_object_tracker` is placed in a queue. When there are some frame drops in the bag, the fluctuation of node `multi_object_tracker` will be significant.

When replacing a bag without frame loss for testing, it was found that the fluctuation of `/perception/object_recognition/tracking/multi_object_tracker/debug/pipeline_latency_ms` was very small, When there were two data in the input data queue of node `multi_object_tracker`,  `/perception/object_recognition/tracking/multi_object_tracker/debug/pipeline_latency_ms` was about 200ms. When there was only one data in the input data queue of node `multi_object_tracker`, `/perception/object_recognition/tracking/multi_object_tracker/debug/pipeline_latency_ms`  was about 100ms. Therefore, the entire perception pipeline took about 100ms from point cloud publishing to tracking completion.

mojomex, 2024-11-22:
@cyn-liu Thank you for the in-depth analysis!

I cannot speak to the latency of the later stages in the pipeline, but we can soon get rid of CropBoxFilter completely as this is being implemented more efficiently in the LiDAR driver ([Nebula](https://github.com/tier4/nebula/)):
* [pr](https://github.com/tier4/nebula/pull/182)
* [docs](https://github.com/mojomex/nebula/blob/d0354087d365b0da67193dd7717d44bdf607c74f/docs/filters.md)

CropBoxFilter will remain as a legacy option, as not all LiDAR vendors are supported by the new feature yet.

closed 2024-12-06 by cyn-liu