autowarefoundation/autoware_universe #8805: planning/control component is dependent on the packages outside of planning/control component

youtalk, 2024-09-08:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

It was discovered in https://github.com/autowarefoundation/autoware/pull/5194 that the planning/control component is dependent on the following packages outside of the planning/control component.

- map/map_loader
- map/autoware_map_projection_loader 
- sensing/autoware_pcl_extensions 
- sensing/autoware_pointcloud_preprocessor

Since the dependencies will no longer follow a parent-child relationship, the components become tightly coupled.

### Purpose

Ensure that there are no dependency packages spanning across components. Connections between components should depend only on the `core` packages or the `autoware.universe/common` packages.

### Possible approaches

Ensure that there are no dependency packages spanning across components. Connections between components should depend only on the `core` packages or the `autoware.universe/common` packages.

### Definition of done

The planning/control packages will no longer directly depend on the following packages.

- [ ] map/map_loader
- [ ] map/autoware_map_projection_loader 
- [ ] sensing/autoware_pcl_extensions 
- [ ] sensing/autoware_pointcloud_preprocessor

takayuki5168, 2024-10-08:
@youtalk 
Thank you for the issues, and sorry to be late.
Could you give me the command to show the list of the packages outside of the planning/control component which the planning/control component is dependent on?

youtalk, 2024-10-11:
@takayuki5168 I only checked by CI builds.
I made a reproducible PR https://github.com/autowarefoundation/autoware/pull/5331 and the latest result was https://github.com/autowarefoundation/autoware/actions/runs/11287273469/job/31392958946?pr=5331#step:5:988.

```
 > [rosdep-universe-planning-control-depend 3/4] RUN rosdep keys --ignore-src --from-paths src     | xargs rosdep resolve --rosdistro humble     | grep -v '^#'     | sed 's/ \+/\n/g'    | sort     > /rosdep-universe-planning-control-depend-packages.txt     && cat /rosdep-universe-planning-control-depend-packages.txt:
1.349 ERROR: no rosdep rule for 'autoware_map_projection_loader'
1.349 ERROR: no rosdep rule for 'autoware_pointcloud_preprocessor'
1.349 ERROR: no rosdep rule for 'map_loader'
```

mitsudome-r, 2024-10-30:
If we are going to include tier4_control_launch and tier4_planning_launch, I had to add the following dependencies as well:
- autoware_control_evaluator
- autoware_evaluator_utils
- autoware_planning_evaluator
- autoware_external_cmd_converter
- autoware_raw_vehicle_cmd_converter

takayuki5168, 2025-01-09:
@youtalk 
https://github.com/autowarefoundation/autoware.universe/pull/9855
By the above PR, the planning and control packages do not now depend on the following packages.
- map/map_loader
- map/autoware_map_projection_loader

youtalk, 2025-01-09:
@takayuki5168 Great work! Thank you for sharing. To also remove `autoware_pointcloud_preprocessor` dependency, please discuss later.

takayuki5168, 2025-02-18:
@youtalk 
Now, the dependency on `autoware_pointcloud_preprocessor` of the planning and control packages is removed as well.