autowarefoundation/autoware_universe #8847: Pointcloud matching fail when vehicle rotates

Joonki0111, 2024-09-11:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

While running Autoware, the pointcloud seems to be acting weird on rotating situation.

https://github.com/user-attachments/assets/28ecc22e-eac2-4fb0-84bd-5ee5988a4bbf

I have no clue what information I have to provide for resolving this issue.

I'm aware that my pointcloud map has very few feature since it's a racing track and this might be the cause of this issue.
If this is the case, is there any possible method to enforce localization?

### Expected behavior

pointcloud keeps aligned with map by NDT scan matcher.

### Actual behavior

mis-matches with map when rotating.

### Steps to reproduce

Will be provided when request is made.

### Versions

Ubuntu 22.04 + ROS2(humble)
Autoware release/v1.0

SakodaShintaro, 2024-09-11:
It would be a good idea to check `ros2 run rqt_runtime_monitor rqt_runtime_monitor` and look at the `ndt_scan_matcher_status`. 
The displayed items may be slightly different from the following picture (due to the Autoware version), but the information should still be helpful. 
First, confirm that `skipping_publish_num` is 0 which means that ndt_scan_matcher publish pose normally.

[image]

Joonki0111, 2024-09-11:
Wow thanks for your quick reply.

https://github.com/user-attachments/assets/060036bf-d01a-4e50-b17b-e3b30b7e9837

I noticed that that execution_time and iteration_num increases while rotation.
[NDT-matching fail -> iteration_num increase -> execution_time increase] is this the case?

SakodaShintaro, 2024-09-11:
I agree that the execution_time and iteration_num are high.

On the other hand, skipping_publish_num remains at 0, so I think the NDT itself is publishing poses successfully.

If the configuration is as per [the recommended architecture](https://autowarefoundation.github.io/autoware-documentation/main/design/autoware-architecture/localization/#recommended-architecture), the NDT poses are processed by the ekf_localizer. My guess is that the ekf_localizer is rejecting the NDT poses for some reason. (timestamp delay or mahalanobis gate, etc.)

Is it possible to see the ekf_localizer status as well?

Joonki0111, 2024-09-11:
Here's the ekf_localizer status.
Also, the architecture of Autoware has no significant change to cause this issue.

https://github.com/user-attachments/assets/3e27225e-9a99-4fcb-99b8-57a1542b2534

SakodaShintaro, 2024-09-11:
It seems that the `ekf_localizer` is working well, so there may be another issue.
To investigate the cause, you can try two approaches.

(1) Run the logging_simulator with only odometry (without the `ndt_scan_matcher`)
This way, you can verify whether the odometry is functioning correctly. (i.e., IMU bias and velocity scaling are properly calibrated).
To acheive this, it's easy to change the output topic name in the `ndt_scan_matcher` config from `/localization/pose_estimator/pose_with_covariance` to `/localization/pose_estimator/pose_with_covariance_tmp`, or something similar.

https://github.com/autowarefoundation/autoware.universe/blob/main/launch/tier4_localization_launch/launch/pose_twist_estimator/ndt_scan_matcher.launch.xml#L3

(May be another file in your version)

(2) Check the trajectories of the `ekf_localizer` and `ndt_scan_matcher`
This method allows you to verify the lidar extrinsic calibration is accurate.
By enabling the pose history checkbox in Rviz(as shown in the next picture), you can visualize each trajectory.

[image]

If the trajectories of the `ndt_scan_matcher` and `ekf_localizer` differ, it could indicate a problem with the lidar-to-base_link calibcation.

Joonki0111, 2024-09-11:
Thanks for your help.
I just checked the trajectories of both of them and seems like there's a problem here.
I'll try something I can do and open this issue again if help is needed :)

closed 2024-09-11 by Joonki0111