autowarefoundation/autoware_universe #8905: Unable to Implement RSS Distance Checking in Scenario Evaluation

brkay54, 2024-09-18:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

## Background

The ODD Working Group has been discussing the implementation of RSS (Responsibility-Sensitive Safety) distance evaluation in running scenarios. The RSS distance, or minimum following distance, depends on dynamic variables such as the current velocity of the ego vehicle and the velocity of the vehicle it's following. However, these variables are not currently accessible in the scenario simulator, posing a challenge for planning evaluation.

## Proposed Solutions

We've explored two potential approaches to address this issue:

### Option 1: Enhance Scenario_Simulator_v2 to Publish Velocities

**Description:** Modify the existing Scenario_Simulator_v2 to publish the required velocity data.

**Pros:**
- Provides more accurate velocity values
- Enables precise RSS distance calculations

**Cons:**
- High implementation cost
- Requires significant changes to the current system

**Status:** Rejected due to resource constraints and implementation complexity.

### Option 2: Integrate RSS Distance Metric into Planning Evaluator

**Description:** Add an RSS distance metric calculation feature to the planning evaluator.

**Pros:**
- More cost-effective solution
- Easier to implement within the current framework
- Allows subscription to RSS distance in the scenario simulator

**Cons:**
- May be less accurate than real-time velocity data

**Status:** Currently the preferred option.

## Next Steps

1. Implement the RSS distance metric in the planning evaluator
2. Create a subscription mechanism in the scenario simulator for the RSS distance
3. Develop a method to check if the distance between Ego and NPC vehicles is below the calculated RSS distance

## Request for Feedback

We welcome additional insights and opinions from the team. Please consider the following questions:

1. Are there any potential drawbacks to Option 2 that we haven't considered?
2. Do you foresee any integration challenges with the current planning evaluator?
3. Are there any alternative approaches we should explore?

Your input is valuable in ensuring we choose the most effective and efficient solution. Please share your thoughts and suggestions in the comments below.

### Purpose

-

### Possible approaches

-

### Definition of done

-

brkay54, 2024-09-18:
cc. @mitsudome-r @Igata-ctrl @kosuke55 @takayuki5168

takayuki5168, 2024-10-02:
@brkay54 
Both options sound good to me. I don't feel any drawbacks and challenges.