autowarefoundation/autoware_universe #8924: Cannot list topics or nodes after launching planning_simulator.launch.xml in Autoware Universe Docker container

batuhanbeytekin, 2024-09-19:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I am facing an issue where, after successfully launching the planning_simulator.launch.xml in the Autoware Universe (CUDA version) Docker container, I am unable to list any ROS 2 topics or nodes using the commands ros2 topic list or ros2 node list. Both commands return no output, even though the simulator itself seems to be functioning normally, and RViz is displayed correctly.

### Expected behavior

I expect to see a list of active topics and nodes after launching the Planning Simulator, allowing me to monitor and interact with the simulation.

### Actual behavior

No output is generated by the ros2 topic list or ros2 node list commands. The Planning Simulator and RViz display correctly, but it seems like the topics and nodes are not available or not properly advertised.

You can see my terminal output from [here](https://github.com/user-attachments/files/17060759/log.txt).

### Steps to reproduce

1. Run the following Docker command to launch the Planning Simulator:
```bash
docker run -it --rm --net=host --gpus all \
  -e DISPLAY=$DISPLAY \
  -v /tmp/.X11-unix/:/tmp/.X11-unix \
  -e XAUTHORITY=/run/user/1000/gdm/Xauthority \
  -e XDG_RUNTIME_DIR=/run/user/1000 \
  -e NVIDIA_DRIVER_CAPABILITIES=all \
  -v /etc/localtime:/etc/localtime:ro \
  -v /home/batu/autoware_map/sample-map-rosbag:/autoware_map:ro \
  -v /dev/dri:/dev/dri \
  ghcr.io/autowarefoundation/autoware:universe-cuda \
  ros2 launch autoware_launch planning_simulator.launch.xml \
  map_path:=/autoware_map \
  vehicle_model:=sample_vehicle \
  sensor_model:=sample_sensor_kit
```
2. Wait for the Planning Simulator and RViz to launch.
3. In another terminal, try listing topics:
```bash
ros2 topic list
```
4. Try listing nodes:
```bash
ros2 node list
```

### Versions

- OS: Ubuntu 22.04
- ROS 2 Distro: Humble
- GPU: NVIDIA GeForce GTX 1660 Ti
- Driver Version: 555.42.02
- Docker Version: 27.1.2
- Autoware Docker Image: ghcr.io/autowarefoundation/autoware:universe-cuda

### Additional context

Any guidance or suggestions on how to resolve this issue would be greatly appreciated!

maxime-clem, 2024-09-20:
I believe the issue is that the network/DDS configuration is not the same between the docker environment (where Autoware and Rviz are running), and your host environment (where you are running `ros2 topic list`).

If that is the case, you have 2 possible solutions:
- Fix the network settings (I think following [these instructions](https://autowarefoundation.github.io/autoware-documentation/main/installation/additional-settings-for-developers/network-configuration/dds-settings/) should work).
- Run the commands from inside the docker container. (`docker exec -it <mycontainer> sh` to start a new shell in the container).

oguzkaganozt, 2025-04-10:
Closing this as it is not possible to reproduce it now.