autowarefoundation/autoware_universe #8940: Add Direct Initialization Option with GNSS pose

meliketanrikulu, 2024-09-23:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

You need to call service to initialize directly now. [Documentation](https://github.com/autowarefoundation/autoware.universe/tree/main/localization/autoware_pose_initializer#direct-initial-position-set)
 However, if you trust the GNSS pose, you may want to initialize directly with the GNSS pose. For example, we think that the pose coming from GNSS for the simulation environment can be used for initialization directly. We think that the feature of using GNSS poses should be added to activate the direct initialization feature for such cases.

### Purpose

In cases where we trust the GNSS pose (simulation tests etc.), we want to shorten the time spent for pose initlizer and ensure that the direct initilization feature is realized through the subscribed GNSS topic.

### Possible approaches

Adding control of gnss poses when the direct initlizer feature is activated may solve the problem.

### Definition of done

The task is completed when Direct initialization is activated using GNSS poses with parameter changes.

SakodaShintaro, 2024-09-25:
I believe the desired behavior can be achieved by commenting out the following line:
https://github.com/autowarefoundation/autoware.universe/blob/d25e9a1bbedc176d4e29c70b3ed9586808bdf813/localization/autoware_pose_initializer/src/pose_initializer_core.cpp#L161
How about adding a parameter to the `autoware_pose_initializer` config that allows toggling whether or not to execute the `align_pose`?

Note: Please make sure that `/sensing/gnss/pose_with_covariance`, which `autoware_pose_initializer` subscribes to, outputs the orientation with the appropriate frame ID (map). If the orientation is an invalid value such as (0, 0, 0, 0), a large number of errors will be output by tf.

(cc. @YamatoAndo  Please let me know if there's anything wrong with my understanding.)

SakodaShintaro, 2024-09-25:
I'm sorry, I missed [this PR](https://github.com/autowarefoundation/autoware.universe/pull/8948).

xmfcx, 2024-09-27:
Let's discuss here:
[Proposal for Simplifying Pose Initialization in Autoware #5285](https://github.com/orgs/autowarefoundation/discussions/5285)

closed 2024-09-27 by xmfcx