autowarefoundation/autoware_universe #8982: fatal error: 'boost/geometry/strategies/agnostic/hull_graham_andrew.hpp' file not found

razr, 2024-09-29:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

`boost/geometry/strategies/agnostic/hull_graham_andrew.hpp` has been removed from `boost` starting from `1.76.0`, please see https://github.com/boostorg/geometry/tree/boost-1.76.0/include/boost/geometry/strategies/agnostic

Maybe find package should be written as:

```cmake
find_package(Boost REQUIRED)
if(Boost_VERSION VERSION_GREATER 107500)
  message(FATAL_ERROR "Boost version must be 1.75 or less")
endif()
``` 

### Expected behavior

see description

### Actual behavior

see description

### Steps to reproduce

see description

### Possible causes

see description

idorobotics, 2024-10-02:
Try posting your question in the Q&A section: https://github.com/orgs/autowarefoundation/discussions/categories/q-a

maxime-clem, 2024-10-07:
@razr fixing the maximum version with CMake will not fix the build issue. Is this okay ?
I think we can check the boost version in the cpp files to determine whether to include the header or not.
```C++
#include <boost/version.hpp>
#if BOOST_VERSION < 107600
  #include <boost/geometry/strategies/agnostic/hull_graham_andrew.hpp>
#endif
```
The call to `boost::geometry::convex_hull` must probably also be conditioned on the `BOOST_VERSION`.

zehranrgi, 2024-11-09:
When I run this code before colcon build 

`rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO
`
ros2 humble downgraded the boost version to 1.74. So, I am not sure how you have boost 1.81? correct me, please.

zehranrgi, 2024-11-09:
I realized that there are two boost version in my case. 
So, I updated the autoware_universe_utils> CMakeList.txt as : 

`
```
cmake_minimum_required(VERSION 3.14)
project(autoware_universe_utils)

include_directories(BEFORE SYSTEM /usr/include)
set(CMAKE_IGNORE_PATH "/usr/local/include")

option(BUILD_EXAMPLES "Build examples" OFF)

find_package(autoware_cmake REQUIRED)
autoware_package()

##find_package(Boost REQUIRED)
find_package(Boost REQUIRED COMPONENTS date_time)

find_package(fmt REQUIRED)

ament_auto_add_library(autoware_universe_utils SHARED
  src/geometry/alt_geometry.cpp
  src/geometry/geometry.cpp
  src/geometry/pose_deviation.cpp
  src/geometry/boost_polygon_utils.cpp
  src/geometry/random_convex_polygon.cpp
  src/geometry/random_concave_polygon.cpp
  src/geometry/gjk_2d.cpp
  src/geometry/sat_2d.cpp
  src/math/sin_table.cpp
  src/math/trigonometry.cpp
  src/ros/msg_operation.cpp
  src/ros/marker_helper.cpp
  src/ros/logger_level_configure.cpp
  src/system/backtrace.cpp
  src/system/time_keeper.cpp
  src/geometry/ear_clipping.cpp
)

target_link_libraries(autoware_universe_utils
  fmt::fmt
)

if(BUILD_TESTING)
  find_package(ament_cmake_ros REQUIRED)

  file(GLOB_RECURSE test_files test/**/*.cpp)

  ament_add_ros_isolated_gtest(test_autoware_universe_utils ${test_files})

  target_link_libraries(test_autoware_universe_utils
    autoware_universe_utils
  )
endif()

if(BUILD_EXAMPLES)
  message(STATUS "Building examples")
  file(GLOB_RECURSE example_files examples/*.cpp)

  foreach(example_file ${example_files})
    get_filename_component(example_name ${example_file} NAME_WE)
    add_executable(${example_name} ${example_file})
    target_link_libraries(${example_name}
      autoware_universe_utils
    )
    install(TARGETS ${example_name}
      DESTINATION lib/${PROJECT_NAME}
    )
  endforeach()
endif()

ament_auto_package()
```

`

then, also, 

`sudo mv /usr/local/include/boost /usr/local/include/boost_backup`

There isn't any issue with this package now, but it took some time to fix it.

razr, 2024-11-09:
@zehranrgi Thank you for looking into it. My OS includes Boost 1.81, which generally works well, except in this specific case. I’ve downgraded to Boost 1.76 and the issue is resolved. So, nothing critical, I just want to point out that there is a dependency in the source code on particular BOOST release.

zehranrgi, 2024-11-11:
Yes, it needs boost 1.76. 

But, also I am getting these errors: 

```
CMake Error at /opt/ros/humble/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Target "test_autoware_universe_utils" links to target "Boost::date_time"
  but the target was not found.  Perhaps a find_package() call is missing for
  an IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  /opt/ros/humble/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/humble/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  /opt/ros/humble/share/ament_cmake_ros/cmake/ament_add_ros_isolated_gtest.cmake:33 (ament_add_gtest)
  CMakeLists.txt:43 (ament_add_ros_isolated_gtest)
```

jiemdev, 2024-11-13:
Hi,
Looks like the class has been moved to boost/geometry/algorithms/detail/convex_hull/graham_andrew.hpp from boost v1.76?

closed 2025-09-24 by razr