autowarefoundation/autoware_universe #9020: distortion_corrector the pointcloud layout is not compatible with PointXYZIRCAEDT

xmfcx, 2024-10-03:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/rosbag-replay-simulation/#steps following this, autoware cannot localize and prints errors.

### Expected behavior

Tutorial works.

### Actual behavior

Fails to localize and prints errors.

```log
1727948014.0067625 [component_container-33] [31m[ERROR 1727948014.005983387] 
[sensing.lidar.right.distortion_corrector_node]: The pointcloud layout is not compatible with PointXYZIRCAEDT.
Aborting (isInputValid() at
/home/mfc/projects/autoware/src/universe/autoware.universe/sensing/autoware_pointcloud_preprocessor/src/distortion_corrector/distortion_corrector.cpp:173)[0m
```

Full log file: https://gist.github.com/xmfcx/1613d133acb87e1a71d9cf57b7fc8381

https://github.com/user-attachments/assets/845b9e53-714f-4326-a47e-b689b705e7ca

### Steps to reproduce

Follow https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/rosbag-replay-simulation/#steps with the latest autoware.

Here is my latest snapshot of problem.repos file: https://gist.github.com/xmfcx/79c4571293336ad39a9f3262ac29ef32

### Versions

Ubuntu 22.04, ROS 2 humble, Autoware latest main 2024-10-03 10:10:37 UTC

### Possible causes

@vividf maybe you might know, a lot of changes were being merged related to the sample_sensor_kit_launch and about autoware_pointcloud_preprocessor package.

xmfcx, 2024-10-03:
Potential culprit:
- https://github.com/autowarefoundation/autoware.universe/pull/8480

xmfcx, 2024-10-03:
So sorry :bow: , I had an outdated AWSIM running in the background, it was fixed when I closed the running simulator :sweat_smile:

vividf, 2024-10-10:
@xmfcx 
I follow the https://autowarefoundation.github.io/AWSIM-Labs/main/GettingStarted/QuickStartDemo/ and install the latest [v1.5.2](https://github.com/autowarefoundation/AWSIM-Labs/releases/tag/v1.5.2), then I also get the same error.

Since you mentioned that it is fixed, I would like to ask how can I fix this. Thanks!

LitoNeo, 2024-10-25:
For AWSIM, the data type of output `/sensing/lidar/top/pointcloud_raw_ex` is **XYZIRCEADT**(as the image shows), but I think what the pointcloud preprocessor need is **XYZIRCAEDT**. I think that is the problem.
[image]

vividf, 2024-10-28:
https://github.com/autowarefoundation/AWSIM-Labs/issues/173

xmfcx, 2024-11-01:
@vividf @LitoNeo could you try the latest release?

I've fixed it in [v1.5.3](https://github.com/autowarefoundation/AWSIM-Labs/releases/tag/v1.5.3) and tested, it worked for me.

- https://github.com/autowarefoundation/AWSIM-Labs/releases

closed 2024-11-04 by xmfcx