autowarefoundation/autoware_universe #9038: Failure in vector mapping due to points having negative xy-coordinates in pointcloud map

wsung1, 2024-10-05:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi, Autoware community,
The mapping site I'm currently working on spans MGRS zones 52S CD and BD.
Because the origin of the map frame is based on the MGRS 52S CD coordinate system, some points in the pointcloud map have negative xy-coordinates.
Based on this pointcloud map, I created a vector map using a bag2lanelet package.
When I loaded the created lanelet into the Vector Map Builder, it appeared correct, but I encountered an error indicating that some points have negative xy-coordinates in the MGRS coordinate system. 
I overlooked this error, as the map seemed plausible. 
However, when I imported the resulting OSM file into Autoware, the lanelet was totally distorted.
The grey line that should encircle the building in the center is missing, as shown below.
[image]
[image]

### Expected behavior

As described above

### Actual behavior

As described above

### Steps to reproduce

As described above

VRichardJP, 2024-10-15:
In your lanelet map OSM file, the `<node>` coordinates are available in 2 different forms: latitude/longitude and MGRS. For example:

```xml
<node id="30" lat="13.801094196677974" lon="100.54001559798192">
    <tag k="mgrs_code" v="47PPR664262"/>
    <tag k="local_x" v="66467.4103"/>
    <tag k="local_y" v="26262.2704"/>
    <tag k="ele" v="-27.514"/>
  </node>
```

You say your map overlaps over 2 MGRS zones, which is likely the cause of issue. For example, Autoware may not check the `mgrs_code` under some conditions while the lanelet map editor will correctly parse it (see https://autowarefoundation.github.io/autoware.universe/main/map/autoware_map_projection_loader/).

VRichardJP, 2024-10-15:
For example if you have set the map projector to MGRS in the `map_projector_info.yaml` file, then the local x/y coordinates are ignored and only each `lat` and `lon` values are used. This projector assumes all points are in the same MGRS zone. If that is not the case, you should see the following warning:

https://github.com/autowarefoundation/autoware_lanelet2_extension/blob/c148815f4483a9acd51bd25abcb26a7a5a99381d/autoware_lanelet2_extension/lib/mgrs_projector.cpp#L63-L68

ll326, 2025-01-09:
I encountered the same problem. Could you please help me with how you solved it? Thank you!

ll326, 2025-01-10:
The error "some points have negative xy-coordinates in MGRS coordinate system" that appears when using the vector map builder software essentially arises because MGRS should have the bottom-left corner of the zone as the origin, but in this software, the origin is set to the vehicle center. This causes a shift in the origin of the constructed map, leading to negative coordinates.

To resolve this origin issue, the origin should be moved to the bottom-left corner of the MGRS zone.

leonidaas, 2025-08-12:
> The error "some points have negative xy-coordinates in MGRS coordinate system" that appears when using the vector map builder software essentially arises because MGRS should have the bottom-left [...]

How to set the origin to the bottom-left? I'm using TIERIV VectorMapBuilder to create the lanelet2 map