autowarefoundation/autoware_universe #9086: autoware_behavior_path_dynamic_obstacle_avoidance_module meets oncoming dynamic obstacles, the drivable area undergoes sudden changes.

luojiaxiang11, 2024-10-15:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

autoware_behavior_path_dynamic_obstacle_avoidance_module meets oncoming dynamic obstacles, the drivable area undergoes sudden changes.

### Expected behavior

When encountering an oncoming dynamic obstacle positioned in the center, dynamic obstacle avoidance should not be carried out.

### Actual behavior

[image]
[image]

https://github.com/user-attachments/assets/aa9fadbf-4ac8-4260-91ed-da2a39151588

### Steps to reproduce

1. Set the starting and ending points in rviz under the simulation mode.
2. Set the oncoming dynamic obstacle in the middle of the planned path.

### Versions

-autoware:main

luojiaxiang11, 2024-10-15:
Can the following conditional statement be removed from the calcMinMaxLateralOffsetToAvoidRegulatedObject function in the file autoware_behavior_path_dynamic_obstacle_avoidance_module/src/scene.cpp？
` if (parameters_->min_front_object_vel < obj_vel)`

shulanbushangshu, 2024-10-15:
> Can the following conditional statement be removed from the calcMinMaxLateralOffsetToAvoidRegulatedObject function in the file autoware_behavior_path_dynamic_obstacle_avoidance_module/src/scene.cpp？ ` if (parameters_->min_front_object_vel < obj_vel)`
@luojiaxiang11 
The condition "parameters_->min_front_object_vel < obj_vel" is to select the moving object.
And in your issue ,The situation "an oncoming dynamic obstacle positioned in the center" should be ignored and This autoware_behavior_path_dynamic_obstacle_avoidance_module  should not work in the sense. Is it right?
 
In this module, I think that if the moving target is considered near the center line, the module shrinking the exercisable area can trigger the speed planning in advance.

luojiaxiang11, 2024-10-15:
> In this module, I think that if the moving target is considered near the center line, the module shrinking the exercisable area can trigger the speed planning in advance.

Yes,i think  autoware_behavior_path_dynamic_obstacle_avoidance_module should not work in that situation. 
when encountering it during real vehicle testing, even though the speed planning can ensure stopping in front of the obstacle, the abrupt change in the drivable area will cause the decision path to fluctuate, leading to steering shaking.

shulanbushangshu, 2024-10-22:
@luojiaxiang11 
Can you plot how much the steering wheel fluctuates?
The current model code has cross-target filtering; For moving targets that are in the center line before detection, I don't think the steering wheel will fluctuate much in a lane with low curvature.

shulanbushangshu, 2024-10-22:
@luojiaxiang11 
Can you plot how much the steering wheel fluctuates?In your video , I can not  see the fluctuates

luojiaxiang11, 2024-10-22:
> @luojiaxiang11 Can you plot how much the steering wheel fluctuates?In your video , I can not see the fluctuates

[image]

shulanbushangshu, 2024-10-31:
Relate to https://github.com/autowarefoundation/autoware.universe/issues/8881
Whether to consider this situation when designing centreline obstacle avoidance @beyzanurkaya