autowarefoundation/autoware_universe #9117: could not set use_intra_process in map.launch.xml

LuZhishan, 2024-10-18:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Default value of `use_intra_process` is `false` [here](https://github.com/autowarefoundation/autoware.universe/blob/ae24913390600d95b3194f7e3ddf43ac823261ce/launch/tier4_map_launch/launch/map.launch.xml#L15),but if set it to `true`, there will be an error

### Expected behavior

load node to `map_container`

### Actual behavior

[image]

### Steps to reproduce

1. set `use_intra_process` to true
2. launch map.launch.xml

### Versions

- OS : ubuntu22.04
- ROS2 : humble
- AutoWare : latest

LuZhishan, 2024-10-18:
Could you help me with this problem ? @YamatoAndo @SakodaShintaro @cyn-liu

YamatoAndo, 2024-10-21:
@LuZhishan 
Sorry, the map component uses [transient_local](https://github.com/autowarefoundation/autoware.universe/blob/main/map/map_loader/src/lanelet2_map_loader/lanelet2_map_loader_node.cpp#L130C75-L130C90), so it may not be possible to set `use_intra_process=true`.

YamatoAndo, 2024-10-22:
@LuZhishan Yes, I think it should be remove.

closed 2024-10-23 by cyn-liu