autowarefoundation/autoware_universe #9131: [autoware_point_types] unable to compile under Jazzy: Could not find PCL

Timple, 2024-10-21:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

```
$ source /opt/ros/jazzy/setup.bash && colcon build --packages-up-to autoware_point_types
Starting >>> autoware_point_types
--- stderr: autoware_point_types
CMake Error at /opt/ros/jazzy/share/pcl_conversions/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package):
  By not providing "FindPCL.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "PCL", but
  CMake did not find one.

  Could not find a package configuration file provided by "PCL" with any of
  the following names:

    PCLConfig.cmake
    pcl-config.cmake

  Add the installation prefix of "PCL" to CMAKE_PREFIX_PATH or set "PCL_DIR"
  to a directory containing one of the above files.  If "PCL" provides a
  separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  /opt/ros/jazzy/share/pcl_conversions/cmake/pcl_conversionsConfig.cmake:41 (include)
  /opt/ros/jazzy/share/pcl_ros/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/jazzy/share/pcl_ros/cmake/pcl_rosConfig.cmake:41 (include)
  /opt/ros/jazzy/share/ament_cmake_auto/cmake/ament_auto_find_build_dependencies.cmake:67 (find_package)
  /opt/ros/jazzy/share/autoware_cmake/cmake/autoware_package.cmake:49 (ament_auto_find_build_dependencies)
  CMakeLists.txt:5 (autoware_package)

---
Failed   <<< autoware_point_types [0.86s, exited with code 1]
```

### Expected behavior

No failed compliation

### Actual behavior

See title and description?

### Steps to reproduce

```Dockerfile
FROM ros:jazzy
SHELL [ "/bin/bash", "-c" ]
RUN apt-get update -qq && apt-get install -qqy python3-colcon-core
RUN mkdir -p /ws/src
WORKDIR /ws
RUN git clone https://github.com/autowarefoundation/autoware.universe /ws/src/autoware.universe
# COPY autoware.universe /ws/src/autoware.universe
RUN rosdep update && rosdep install --from-paths src/autoware.universe/common/autoware_point_types --ignore-src -y
RUN source /opt/ros/${ROS_DISTRO}/setup.bash && colcon build --packages-up-to autoware_point_types
```

### Versions

- ROS 2: Jazzy
- autoware: main (5a89973fb7aa94d37488144b933324e0860d675b)

### Possible causes

Changes in dependencies from humble to jazzy?

knzo25, 2024-10-22:
@Timple 
Does it fail if you add `<depend>libpcl-all-dev</depend>` to the package.xml?

Btw, we do not support Jazzy as of now :smiling_face_with_tear:

knzo25, 2024-10-22:
@Timple 
I see that you found the reason in the referenced issues.
Would you like us to add a temporary dependency to the package? or wait until the fixes are merged upstream? Should the issue be closed?

Btw, we do not support Jazzy as of now :smiling_face_with_tear:

Timple, 2024-10-22:
I think I did find the root cause. No need for a temporary solution, bit thank you for offering!

closed 2024-10-22 by Timple

patham9, 2024-10-26:
@Timple : You found the root cause? I got the same issue, please mention the solution as else the issue shouldn't be closed.

Timple, 2024-10-26:
Root cause was here: https://github.com/ros-perception/perception_pcl/pull/447

We have to wait for a new release of perception_pcl