autowarefoundation/autoware_universe #9146: Too many nodes with same name

LuZhishan, 2024-10-24:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

When I launch many yolox node with `multiple_yolox.launch.xml` [here](https://github.com/autowarefoundation/autoware.universe/blob/main/perception/autoware_tensorrt_yolox/launch/multiple_yolox.launch.xml), there will be many node with same name

### Expected behavior

There should be many node with different names

### Actual behavior

[image]

### Steps to reproduce

1. change the [image_number](https://github.com/autowarefoundation/autoware.universe/blob/8e0d40b8510ff69c4653842433dd4d4b284beb79/perception/autoware_tensorrt_yolox/launch/multiple_yolox.launch.xml#L10C14-L10C26) larger than 1
2. launch `multiple_yolox.launch.xml`

### Versions

- OS: ubuntu22.04
- ROS2: hubmle
- AutoWare: latest

### Additional context

There are two ways to solve this problem
The first one, change node name of [tensorrt_yolox](https://github.com/autowarefoundation/autoware.universe/blob/8e0d40b8510ff69c4653842433dd4d4b284beb79/perception/autoware_tensorrt_yolox/launch/yolox_s_plus_opt.launch.xml#L24C85-L24C99) to `$(anon tensorrt_yolox)`
The second one, add a param passed form uper launch named `tensorrt_yolox_1` for example.

cyn-liu, 2024-10-25:
@LuZhishan Thanks for your suggestion.

badai-nguyen, 2024-11-08:
> @LuZhishan Thanks for your suggestion. When I used the first method to solve this problem([draft PR](https://github.com/autowarefoundation/autoware.universe/pull/9156)), I found that the node names were still the same. I think this is abnormal. Does anyone know the reason? @mitsudome-r @xmfcx @dan-dnn @ktro2828 @manato
> 
> ```
> #ros2 node list
> WARNING: Be aware that are nodes in the graph that share an exact name, this can have unintended side effects.
> /image_transport_decompressor_node_cynthia_liu_25507_8885756925087757029
> /image_transport_decompressor_node_cynthia_liu_25507_8885756925087757029
> /tensorrt_yolox_cynthia_liu_25507_5677783873391810063
> /tensorrt_yolox_cynthia_liu_25507_5677783873391810063
> ```
@cyn-liu 
I guess the reason is the `multiple_yolox.launch.xml` launch multi time yolox.launch.xml with `name="$(anon tensorrt_yolox)"` but in a very short time so that ROS2 is fail to create the distinct identifiers for each instance? 
I think a better solution is adding a parameter for node name in yolox.lauch.xml and explicitly assign them in `multi_yolox.launch.xml`.

amadeuszsz, 2024-11-08:
@badai-nguyen

> I think a better solution is adding a parameter for node name in yolox.lauch.xml and explicitly assign them in `multi_yolox.launch.xml`.

FYI recently I was implementing something similar and I just followed same way as it is in ROS 2 [geometry2](https://github.com/ros2/geometry2/blob/d0dbd516d1748cfe1b293e0f7d5b2f82ddd3438a/tf2_ros/src/transform_listener.cpp#L47-L51) packages. It doesn't look nice but it works. I assume this approach might be good for packages with multiple instances.

cyn-liu, 2024-11-22:
> @badai-nguyen
>
>
> FYI recently I was implementing something similar and I just followed same way as it is in ROS 2 [...]

Perhaps method ROS 2 [geometry2](https://github.com/ros2/geometry2/blob/d0dbd516d1748cfe1b293e0f7d5b2f82ddd3438a/tf2_ros/src/transform_listener.cpp#L47-L51) packages is effective, but it is not the best choice in this node.  In node [tensorrt_yolox](https://github.com/cyn-liu/autoware.universe/blob/248bba7f2a3cd6a8d0777350c6dce062e79f5967/perception/autoware_tensorrt_yolox/src/tensorrt_yolox_node.cpp#L30), object `node_options` is const type, and cannot call non const member function `arguments()`.

closed 2025-01-14 by mitsudome-r