autowarefoundation/autoware_universe #9183: Unable to click the Auto button after selecting the starting and ending points during  planning simulation

cyn-liu, 2024-10-30:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

When I use the latest Autoware code for planning simulation, even after planning the path, the Auto button remained disabled and could not be clicked.
Start command:
`ros2 launch autoware_launch planning_simulator.launch.xml map_path:=./banqiao-new/`
console output:
```bash
[rviz2-45] [INFO] [1730257751.816463405] [rviz2]: Setting estimate pose: Frame:map, Position(51.1698, -15.0847, 0), Orientation(0, 0, -0.160182, 0.987087) = Angle: -0.32175
[initial_pose_adaptor_node-42] [INFO] [1730257751.816635655] [default_ad_api.helpers.initial_pose_adaptor]: fit_target: vector_map, frame: map
[component_container_mt-38] [INFO] [1730257751.840882003] [control.trajectory_follower.lane_departure_checker_node]: waiting for route msg...
[autoware_scenario_selector_node-29] [INFO] [1730257751.868930471] [planning.scenario_planning.scenario_selector]: Waiting for route.
[planning_evaluator-36] [WARN] [1730257751.886723038] [route_handler]: setMap() for invalid version map: 
[component_container_mt-38] [INFO] [1730257752.063924237] [control.trajectory_follower.controller_node_exe]: Waiting for trajectory data
[component_container_mt-38] [INFO] [1730257752.063988614] [control.trajectory_follower.controller_node_exe]: Control is skipped since input data is not ready.
[component_container_mt-32] [INFO] [1730257752.713678462] [planning.scenario_planning.lane_driving.behavior_planning.behavior_path_planner]: waiting for scenario_topic
[topic_state_monitor_node-12] [INFO] [1730257753.778787103] [system.topic_state_monitor_trajectory_follower_control_cmd]: /control/trajectory_follower/control_cmd has not received. Set ERROR in diagnostics.
[topic_state_monitor_node-11] [INFO] [1730257753.819254928] [system.topic_state_monitor_scenario_planning_trajectory]: /planning/scenario_planning/trajectory has not received. Set ERROR in diagnostics.
[topic_state_monitor_node-13] [INFO] [1730257753.982490075] [system.topic_state_monitor_control_command_control_cmd]: /control/command/control_cmd has not received. Set ERROR in diagnostics.
[topic_state_monitor_node-10] [INFO] [1730257754.043716689] [system.topic_state_monitor_mission_planning_route]: /planning/mission_planning/route has not received. Set ERROR in diagnostics.
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/planning/scenario_planning/parking/costmap_generator' in container '/planning/scenario_planning/parking/parking_container'
[component_container_mt-34] [INFO] [1730257756.134840728] [planning.scenario_planning.parking.parking_container]: Load Library: /home/lc/workspace/humble/autoware_20241023/install/autoware_freespace_planner/lib/libfreespace_planner_node.so
[component_container_mt-34] [INFO] [1730257756.140237184] [planning.scenario_planning.parking.parking_container]: Found class: rclcpp_components::NodeFactoryTemplate<autoware::freespace_planner::FreespacePlannerNode>
[component_container_mt-34] [INFO] [1730257756.140277397] [planning.scenario_planning.parking.parking_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<autoware::freespace_planner::FreespacePlannerNode>
[component_container_mt-34] [INFO] [1730257756.186496233] [planning.scenario_planning.parking.freespace_planner]: initialize planning algorithm: astar
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/planning/scenario_planning/parking/freespace_planner' in container '/planning/scenario_planning/parking/parking_container'
[component_container_mt-34] [INFO] [1730257756.205628000] [planning.scenario_planning.parking.parking_container]: Load Library: /home/lc/workspace/humble/autoware_20241023/install/glog_component/lib/libglog_component.so
[component_container_mt-34] [INFO] [1730257756.207681080] [planning.scenario_planning.parking.parking_container]: Found class: rclcpp_components::NodeFactoryTemplate<GlogComponent>
[component_container_mt-34] [INFO] [1730257756.207708186] [planning.scenario_planning.parking.parking_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<GlogComponent>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/planning/scenario_planning/parking/glog_component' in container '/planning/scenario_planning/parking/parking_container'
[topic_state_monitor_node-12] [INFO] [1730257756.878767074] [system.topic_state_monitor_trajectory_follower_control_cmd]: /control/trajectory_follower/control_cmd has not received. Set ERROR in diagnostics.
[topic_state_monitor_node-11] [INFO] [1730257756.919227213] [system.topic_state_monitor_scenario_planning_trajectory]: /planning/scenario_planning/trajectory has not received. Set ERROR in diagnostics.
[component_container_mt-38] [INFO] [1730257756.940715258] [control.trajectory_follower.lane_departure_checker_node]: waiting for route msg...
[autoware_scenario_selector_node-29] [INFO] [1730257756.968900456] [planning.scenario_planning.scenario_selector]: Waiting for route.
[topic_state_monitor_node-13] [INFO] [1730257756.985936131] [system.topic_state_monitor_control_command_control_cmd]: /control/command/control_cmd has not received. Set ERROR in diagnostics.
[topic_state_monitor_node-10] [INFO] [1730257757.043721767] [system.topic_state_monitor_mission_planning_route]: /planning/mission_planning/route has not received. Set ERROR in diagnostics.
[component_container_mt-38] [INFO] [1730257757.073942627] [control.trajectory_follower.controller_node_exe]: Waiting for trajectory data
[component_container_mt-38] [INFO] [1730257757.074016762] [control.trajectory_follower.controller_node_exe]: Control is skipped since input data is not ready.
[component_container_mt-32] [INFO] [1730257757.714020277] [planning.scenario_planning.lane_driving.behavior_planning.behavior_path_planner]: waiting for scenario_topic
[rviz2-45] [INFO] [1730257758.488420394] [rviz2]: Setting goal pose: Frame:map, Position(255.535, -92.384, 0), Orientation(0, 0, -0.22534, 0.97428) = Angle: -0.454585
[component_container_mt-27] [INFO] [1730257758.756907583] [route_handler]: getMainLanelets: lanelet_sequence = [ids: 110 ]
[component_container_mt-27] [INFO] [1730257758.758384666] [planning.mission_planning.mission_planner]: initial pose - x: 51.169815, y: -15.084711, z: -2.108893
[component_container_mt-27] [INFO] [1730257758.758415891] [planning.mission_planning.mission_planner]: initial orientation - qx: 0.000141, qy: 0.000870, qz: -0.160182, qw: 0.987087
[component_container_mt-27] [INFO] [1730257758.758424795] [planning.mission_planning.mission_planner]: goal pose - x: 255.534637, y: -92.383980, z: 0.000000
[component_container_mt-27] [INFO] [1730257758.758431336] [planning.mission_planning.mission_planner]: goal orientation - qx: 0.000000, qy: 0.000000, qz: -0.225340, qw: 0.974280
[component_container_mt-32] [WARN] [1730257758.816564596] [route_handler]: setMap() for invalid version map: 
[component_container_mt-32] [WARN] [1730257758.829397410] [route_handler]: setMap() for invalid version map: 
[component_container_mt-38] [INFO] [1730257758.841190313] [control.trajectory_follower.lane_departure_checker_node]: waiting for reference_trajectory msg...
[component_container_mt-33] [WARN] [1730257758.877619068] [route_handler]: setMap() for invalid version map: 
```

[image]

### Expected behavior

Auto button enabled, clicking the auto button allows the EGO to run along the planned path.

### Actual behavior

 Auto button remained disabled and could not be clicked.

### Steps to reproduce

None

### Versions

- OS ubuntu22.04  
- ROS humble
- Autoware latest

### Possible causes

None

### Additional context

None

maxime-clem, 2024-10-30:
I cannot reproduce the issue. Can you share the full log ? The log messages you shared are from before the path was published so the messages are not enough to understand the issue.

cyn-liu, 2024-10-30:
@maxime-clem Thank you for your reply. My planning simulation can now run normally. Due to I didn't set `ROS_DOMAIN_ID`, other people's ROS messages have affected me.

closed 2024-10-30 by cyn-liu