autowarefoundation/autoware_universe #9210: Static avoidance planner generates wrong trajectory when tries to avoid obstacles on two sides

firemark, 2024-10-31:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

When the two obstacles are on two sides around narrow lanelet and one obstacle  is "parked" then generates wrong trajectory that triggers `obstacle_stop` by the obstacle cruiser planner.

### Preview

https://github.com/user-attachments/assets/9d1bf772-f940-4682-af5f-8ce02bfe3c6b

### Expected behavior

Due to sufficient space for both sides, trajectory should be not modified.

### Actual behavior

The avoidance planner generated trajectory too close to the obstacle that triggers `obstacle_stop` from the obstacle cruise planner.

### Steps to reproduce

I tested on carla in Town03.

I generated two vehicles on both sides around lanelet.

The first vehicle should collide with the lanelet (it should be tagged as _parked_ vehicle). 

The second should be on the adjacent lanelet.

### Versions

- OS: Ubuntu 24.04
- ROS: humble (docker image)
- Autoware commit id: a967adbad7a585aa7a211422ff6a6f635e57196c

### Possible causes

What I understand:

1. To avoid an obstacle, it's generated _shift points_  as single _shift line_
2. For both sides, they're generated two _shift lines_ (negative and positive) and merged into one.
3. Due to some reasons (e.g. not sufficient space for _hard margin__ or an obstacle is near goal/vehicle) generating _shift points_ are discarded for the obstacle (or side?)

#### Case A

Two _shift lines_ are generated properly.

[image]

#### Case B

Due to not sufficient space for the obstacle on the left side, _shift line_ for left side is not generated properly and merged _shift line_ is equal to right _shift line_. This generates wrong trajectory which is too near the left obstacle.

[image]

satoshi-ota, 2024-11-26:
@firemark Thank you for you bug report.
In current implementation, the [static_obstacle_avoidance](https://github.com/autowarefoundation/autoware.universe/tree/main/planning/behavior_path_planner/autoware_behavior_path_static_obstacle_avoidance_module) module doesn't support fine adjustment in such a case explicitly.

Btw, have you tried using this module...? You can enable this module from **[this](https://github.com/autowarefoundation/autoware_launch/blob/e14b2c849950930475fc644b78e62630f2510cd7/autoware_launch/config/planning/preset/default_preset.yaml#L9-L11)** flag.
This module cut drivable area based on object position and footprint, and [this](https://github.com/tier4/autoware.universe/tree/tier4/main/planning/autoware_path_optimizer) module push the path into the modified drivable area. The latter one uses optimization logic, so it's good at fine adjustment.

Hopefully, I think it can pass through between the two obstacles.

firemark, 2025-02-04:
@satoshi-ota Sorry for a late response.

Currently I haven't tried and for my current project it's not a big deal for the case. I found the case in simulation and I think is just a bug. Thanks for tips how to resolve the problem.