autowarefoundation/autoware_universe #9248: No objects is detected using the new TransFusion-L lidar detection

vincentcheng62, 2024-11-06:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

By setting "lidar_detection_model" to "transfusion" in the respective config files, transfusion can run,
however no objects is detected (in a specific scene in carla that has an obvious objects)

When i do 
ros2 topic echo /perception/object_recognition/detection/transfusion/objects

I got 
---
header:
  stamp:
    sec: 4135
    nanosec: 243743563
  frame_id: base_link
objects: []
---
header:
  stamp:
    sec: 4135
    nanosec: 343743565
  frame_id: base_link
objects: []
---
header:
  stamp:
    sec: 4135
    nanosec: 443743566
  frame_id: base_link
objects: []
---

### Expected behavior

"objects" field should not be empty [] when i do 
ros2 topic echo /perception/object_recognition/detection/transfusion/objects

### Actual behavior

"objects" field are all empty [] when i do 
ros2 topic echo /perception/object_recognition/detection/transfusion/objects

### Steps to reproduce

1. Download the transfusion.onnx pre-trained weights from autoware official website
2. setting "lidar_detection_model" to "transfusion" in the respective config files
3. Run Carla and scenario runner to generate some vehicle and pedestrian objects

### Versions

-OS: Ubuntu 22.04 LTS
-ROS2: Humble
-Autoware universe (latest version in Sep)

knzo25, 2024-11-06:
@amadeuszsz 
Do you know anything about this?

amadeuszsz, 2024-11-06:
@vincentcheng62 
TransFusion needs intensity field, please check if your input PointCloud2 topic has intensity field. If intensity field exist, please check values of intensities. You can do it in PointCloud2 visualization settings in rviz. If intensities range are from 0 to 0, your data is lack of intensity, correct value should be from 0 to 255.

When you confirm correctness of input point cloud,  could you please share rosbag for us with PointCloud2 topic?

vincentcheng62, 2024-11-07:
Thanks @amadeuszsz 

I have checked the correctness of the pointcloud published by CARLA ros bridge, which gives intensity field from 0 to 255, refer to the attached screen shot image
[image]
[image]

Attached a 10sec ROS2 bag that contains topic '/sensing/lidar/concatenated/pointcloud' in which at coordinate (x,y)=(18,0) there is a simulated vehicle.

amadeuszsz, 2024-11-07:
@vincentcheng62 
As you can see on your picture, this intensity does not make any sense. It look like intensity values are assigned based only on distance. For the instance, in your rosbag to see intensity variation, you can uncheck *Autocompute Intensity Bounds* and see that all intensities are between 208-249 range. Setting bounds to 0-255 will give you a preview for what TransFusion sees.

[image]

First, check if direct point cloud from Carla has same issue - if intensities looks rational, it will indicate a bug in our pointcloud preprocessor. In that case, please share a rosbag with all raw point clouds from carla.

If Carla's point cloud itself does not provide any valuable information about intensities, TransFusion might not work as expected. And indeed, looking into Carla's [calculations](https://github.com/carla-simulator/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/RayCastLidar.cpp#L93-L104), the intensity is simply an exponential with exponent distance times constant factor. To make it works, Carla's map needs to consider reflection values for each texture, and then calculate intensity base on this reflection, distance, angle of incidence, weather conditions and other constants.

vincentcheng62, 2024-11-07:
Yes u r right, so the 3D lidar intensity value must be meaningful in order to make Transfusion algorithm to work properly.

I recheck CARLA documentation and source code, it seems that the simulated lidar intensity value is somewhat "faked" so that it does not represent the case in reality.

I run the autoware universe official rosbag again in which the data is real data, Transfusion is okay now.

amadeuszsz, 2024-11-08:
Closing as solved.

closed 2024-11-08 by amadeuszsz