autowarefoundation/autoware_universe #9282: Concatenate node with main LiDAR concept

vividf, 2024-11-11:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

[This PR](https://github.com/autowarefoundation/autoware.universe/pull/8300) brings the new concatenate algorithm. 

There are requests for vehicles using the top LiDAR as the 'main' LiDAR. If the top LiDAR is not concatenated with the other LiDARs, it may affect certain modules within the perception stack.

### Purpose

Provide a more robust algorithm for perception stack that needs the concept of main LiDAR.

### Possible approaches

- First option: Implement additional parameters and logic in the concatenate node, allowing users to select the main LiDAR. If the main LiDAR is not present in the concatenated output, the output should be discarded.

- Second option: Implement an additional function in the nodes that require the 'main' LiDAR concept. This function should analyze diagnostic messages from the concatenate node to determine whether to discard the point cloud.

### Definition of done

Implement one of the solution that describe above

felixf4xu, 2025-03-17:
#8300 is merged and this issue should be closed to make the issue list clean. :-)

vividf, 2025-03-17:
In the final decision, we chose not to implement the main LiDAR concept in the concatenation node. Nodes requiring this concept should implement the additional feature in their own code.

vividf, 2025-03-17:
@felixf4xu This is actually the feature that needs to be implemented after #8300.
But thanks for reminding me of this issue.

closed 2025-03-17 by vividf