autowarefoundation/autoware_universe #9323: param file is incorrect

ZahraBoroujeni, 2024-11-14:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

In the launch file, please correct the path of the  bytetrack_visualizer_param_path:
```
<arg name="bytetrack_visualizer_param_path" default="$(find-pkg-share autoware_bytetrack)/config/bytetrack_visualizer.param.yaml"/>
```

### Expected behavior

to link to the bytetrack_visualizer.param.yaml instead of bytetrack.param.yaml

### Actual behavior

link to bytetrack.param.yaml for bytetrack_visualizer node

### Steps to reproduce

Run the launch file with desired parameters for visualizer.

ZahraBoroujeni, 2024-11-14:
https://github.com/autowarefoundation/autoware.universe/blob/main/perception/autoware_bytetrack/launch/bytetrack.launch.xml#L8

ktro2828, 2024-11-27:
@ZahraBoroujeni Thank you for your report, and sorry for the late response.
I've created a PR to address this PR, please take look on https://github.com/autowarefoundation/autoware.universe/pull/9490.

closed 2024-12-02 by ktro2828