autowarefoundation/autoware_universe #9599: NDT Scan matching takes too long to execute

shin-ka, 2024-12-09:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I am using the standalone NDT scan matcher package to localize in an indoor environment simulation. I am using a 16 channel lidar. The initial pose estimate is the same as the current groundtruth pose coming from the simulation and the robot is stationary. Upon configuring the param file and launch file and loading the offline map (again generated using 16 channel lidar), I am seeing that NDT is taking too long to execute, approximately ~5000 ms. This is the behaviour in the ideal scenario where the scans are already aligned and the predicted ndt_pose is actually same as my initial guess given.

### Expected behavior

I am expecting the scan alignment to work faster and not take 5 seconds.

### Actual behavior

It is taking ~5000 ms to predict the ndt_pose.

Here the groundtruth pose is the red marker and initial guess is yellow marker. The current lidar scan is also shown.
[image]

This is the result after NDT matching. Green marker is the estimated pose and purple points correspond to the loaded offline map.
[image]

### Steps to reproduce

1. The global frame of reference in my project is 'odom'. The offline map has also been saved in the odom frame.
2. The config file that I am using : 
[ndt_scan_matcher.param.txt](https://github.com/user-attachments/files/18062075/ndt_scan_matcher.param.txt)

### Versions

- OS: Ubuntu 22.04
- ROS 2 : Humble

YamatoAndo, 2024-12-09:
@shin-ka Are you building with the release option?
`colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release`

SakodaShintaro, 2024-12-10:
(1) If the map contains too many points, the NDT execution time can become significantly slow.
  (1-a) How many points are in your map?
  (1-b) Have you tried dividing your map using [pointcloud_divider](https://github.com/autowarefoundation/autoware_tools/tree/main/map/autoware_pointcloud_divider)?

(2) Additionally, if the map file format is in ASCII (even though this is a rare case), the NDT execution time tends to slow down. Using a binary format is generally recommended for better performance. Could you check this?

shin-ka, 2024-12-10:
> (1) If the map contains too many points, the NDT execution time can become significantly slow. (1-a) How many points are in your map? (1-b) Have you tried dividing your map using [...]

(1-a) The offline map has 658 points currently.
(1-b) My map is not very huge, so not dividing the map currently. But I will try this.

(2) Could you tell me how to check which format the map file is in?

YamatoAndo, 2024-12-10:
Here are some possible causes that come to mind:

- Low computer performance

  If this is a topic about execution speed, please make sure to share your computer's specifications.
Additionally, for comparison purposes, let us know the execution time when [running the sample dataset](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/rosbag-replay-simulation/) with Autoware v.39.0.

- Unintended changes to the source code

  What exactly is "the standalone NDT scan matcher package"?
For example, have you made any implementations where the map point cloud is loaded in every loop?
If possible, please share the source code.

- Slow input LiDAR point cloud frequency

  Check whether the input LiDAR point clouds to the NDT scan matcher package are delayed.

shin-ka, 2024-12-10:
1. Low computer performance
Here are the specs of the laptop on which I'm running the package:
HP Omen equipped with an 13th Gen Intel Core i7-13700HX × 24, 16 GB RAM, NVIDIA RTX 4050 GPU

Since I am not using the entire autoware stack, will be still possible for me to get results from this bag?

2. Unintended changes to the source code
I am using only this[https://github.com/autowarefoundation/autoware.universe/tree/humble/localization/autoware_ndt_scan_matcher](url) and this [https://github.com/autowarefoundation/autoware.universe/tree/humble/map/autoware_map_loader](url) package from the Autoware stack in my project. These packages are being used as it is, with the only changes being in the topic names in launch files and the ndt_scan_matcher param file (that I shared above).

3. Slow input LiDAR point cloud frequency
The lidar data from my simulator is being published at ~20 Hz.

Update: I bumped up the frequency at which I am getting initial pose from my simulator to 40 Hz and I am using 8 threads. Now the ndt_poses are coming at a frequency almost equivalent to lidar data, at ~ 16-18 Hz.

YamatoAndo, 2024-12-10:
> Update: I bumped up the frequency at which I am getting initial pose from my simulator to 40 Hz (earlier it was 10 Hz) and I am using 8 threads. Now the ndt_poses are coming at a frequency almost equivalent to lidar data, at ~ 16-18 Hz.

So, has your problem been resolved?

shin-ka, 2024-12-10:
One last question, will the frequency at which I get ndt_poses depend on the frequency of lidar data?

YamatoAndo, 2024-12-10:
Yes,
If there is no processing delay, it will be equivalent to the frequency of the LiDAR data.

shin-ka, 2024-12-10:
Thank you for your patience and quick responses.
I will close the issue now.

closed 2024-12-10 by shin-ka