autowarefoundation/autoware_universe #9614: How can i export custom transfusion.pth to transfusion.onxx which trained by openpcdetv0.6 or mmdetection3D?

pengboyemo, 2024-12-11:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

autoware_lidar_transfusion have transfusion.onnx model, I use openpcdet_v0.6 to train transfusion, but how can i export the custom pth to onnx, can you share the script?

### Purpose

Export the transfusion custom pth to onnx which is used for inference.

### Possible approaches

None

### Definition of done

None

amadeuszsz, 2024-12-11:
Hi @pengboyemo 
Currently, we work intensively in order open source our framework for ML training and deployment. Nevertheless, deployment script has to be suited to specific dataset, therefore I suggest you to check how it is done for [mmdetection3d](https://github.com/XuyangBai/TransFusion/tree/master) original training loop implementation and [deployment script](https://github.com/open-mmlab/mmdeploy/blob/main/tools/deploy.py).

pengboyemo, 2024-12-12:
Hi @amadeuszsz , thanks for your response, i will try coding deployment script with my custom dataset and configs, and meantime i am looking forward of your ML framework for training and deployment as soon as possible.

closed 2024-12-12 by pengboyemo