autowarefoundation/autoware_universe #9666: Cannot reach to the goal at the corner.

Kim-mins, 2024-12-17:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi team,

I'm currently running Autoware with Carla, and found a situation that the vehicle cannot reach to the goal at the corner.
Here's the rviz video of the situation: [[rviz](https://salbox.synology.me/drive/d/s/11MdXFZTggpXLKJdckskEl4vED6Nkx1a/JZmtqSUkKlN3IvzS-CH_P605amnE-nJJ-wLgAvMdx5ws)]

As the video shows, the goal point is at the corner and the vehicle stops before reaching to the goal.

I partially checked the control command(from `/control/command/control_cmd`) published in last few seconds, and I could get the figure below: (x: # of published data, y: velocity/acceleration)
[image]
And it shows the velocity and the acceleration at the end of the driving is `0` and below `0`, respectively.

For better debugging, here's more information: 
- About starting and end points (from launch.log):
```
initial pose - x: 145.509345, y: -133.451777, z: 0.069356
initial orientation - qx: -0.000332, qy: -0.000049, qz: -0.000555, qw: 1.000000
goal pose - x: 342.991272, y: -2.307267, z: 2.000000
goal orientation - qx: 0.000000, qy: 0.000000, qz: 0.147707, qw: 0.989031
```
- Map: Town01 [[Town01](https://drive.google.com/drive/folders/1bZfKbq6H9dF1DveuW7iIJiJKN0-W5l03)]
- ros2bag file: [[ros2bag](https://salbox.synology.me/drive/d/s/11McxHAUk1nW2gZSlg1iOVdify9NxTpd/eFYq7DkZ8fG5zvLxbP506wdVTL3wlom6-tbsgGAFz5ws)] 
- launch.log file: [[launch.log](https://salbox.synology.me/drive/d/s/11MdEAPpYToNVjW8ITurgocU5hXgVeiK/3IMk3tUggFrmV3QqtmyFIze4O0XPsTAr-3LtADAxz5ws)]

Can anyone please check this? Thanks!

### Expected behavior

I hope the vehicle to reach to the goal point.

### Actual behavior

But the vehicle could not reach to the goal point, and stops before the goal.

### Steps to reproduce

1. Pull image from `https://github.com/autowarefoundation/autoware/pkgs/container/autoware/321564631?tag=universe-devel-cuda-0.39.1-amd64` and run Autoware with Carla, using [autoware_carla_interface](https://github.com/autowarefoundation/autoware.universe/tree/0.39.0/simulator/autoware_carla_interface).
2. Run ros2bag file above.

### Versions

- OS: Ubuntu 22.04 (Docker), Ubuntu 20.04 (host)
- Autoware: [0.39.1](https://github.com/autowarefoundation/autoware/tree/0.39.1)
- Carla: 0.9.15
- autoware_carla_interface: [link](https://github.com/autowarefoundation/autoware.universe/tree/0.39.0/simulator/autoware_carla_interface)

maxime-clem, 2024-12-17:
The emergency stop activates and causes the early stops.
I am not sure what triggers the emergency stops, and I am not sure why the vehicle does not restart at the end.

Kim-mins, 2024-12-17:
Thank you for the response @maxime-clem!

Actually, I have no idea for debugging. Could you please provide me some recommendations for better investigation?

Thanks!

maxime-clem, 2024-12-17:
I found why the vehicle does not restart. It is because when ego last stops (because of the emergency stop), it is around 40cm away from the next trajectory stop points. In the parameters, it is defined that the longitudinal controller node transitions to the `STOPPED` state when within 49cm of the next stop point (https://github.com/autowarefoundation/autoware_launch/blob/main/autoware_launch/config/control/trajectory_follower/longitudinal/pid.param.yaml#L14C5-L14C29).

Kim-mins, 2024-12-17:
Thank you for the investigation @maxime-clem!

I read your comment, and maybe I should shorten the `stopping_state_stop_dist` threshold value from  `0.49` to `0.40`(or less than `0.40`). Could you please check if I'm doing right?

maxime-clem, 2024-12-17:
Reducing the value of this parameter would allow the ego vehicle to restart yes.
But here the main issue is the emergency stop. I will investigate it and I suspect it is the same issue as https://github.com/autowarefoundation/autoware.universe/issues/9672.

Kim-mins, 2024-12-17:
Thank you for the response @maxime-clem!

I tested after shortening the distance, and the vehicle could move after the emergency stop. So the error seems gone after your suggested patch! Thanks!

For the emergency stop issue, I just checked your answer in #9672. Can I follow the workaround you suggested there?

Thanks a lot!

maxime-clem, 2024-12-17:
> For the emergency stop issue, I just checked your answer in https://github.com/autowarefoundation/autoware.universe/issues/9672. Can I follow the workaround you suggested there?

Yes it looks to be the same issue.

closed 2024-12-19 by idorobotics