autowarefoundation/autoware_universe #9673: Cannot make proper acceleration at uphill.

Kim-mins, 2024-12-17:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi team,

I'm currently running Autoware with Carla, and found a situation that the vehicle stucks at uphill.
Here's the rviz videos of the situation: [[rviz1](https://salbox.synology.me/drive/d/s/11MppuIWhMYs4vTIVcWtT1Tu7rmbXa6g/AIatw9bqHVlhMJ41TZXd-RK2_TGoq2FI-Qr5g7pqd5ws)] [[rviz2](https://salbox.synology.me/drive/d/s/11MrDaIG5NWRGGdRT4g6D7J8vgKpMXHV/XD7gU2T0ZLktvYv2XseR4P99IZ8gF1UE-qbQgeI2f5ws)]

- rviz1: This video shows a driving with default setting of [slope compensation](https://autowarefoundation.github.io/autoware.universe/main/control/autoware_pid_longitudinal_controller/#slope-compensation_1), and it shows that the vehicle cannot go uphill.
- rviz2: I tested the situation with [planning simulator](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/), and this also shows that the vehicle cannot go uphill either. I checked this because I thought this issue could be from bad localization at the uphill, but it seems not (at least for now).

Also, while running the simulation above, I cannot know why the shape of the drivable area is weird.
[image]

Here's more information of the situation:
- Starting/Goal point (from launch.log)
```
initial pose - x: -74.346236, y: 17.611892, z: -0.062959
initial orientation - qx: -0.004039, qy: 0.003657, qz: 0.707924, qw: 0.706268
goal pose - x: -1.848311, y: 87.232613, z: 2.275307
goal orientation - qx: 0.000000, qy: 0.000000, qz: -0.719365, qw: 0.694632
```
- Map: Town03 [[Town03](https://drive.google.com/drive/folders/1rt1pKEfY-hqC2pMrHfajg1EVfjlIcAX5)]
- ros2bag file: [[ros2bag](https://salbox.synology.me/drive/d/s/11MpdAmgOMq56Ig3PHN9dCxX9gevdxdh/Hf_X_lznGsAWgfwV2vXhV1xAnUtB6-Hy-obAg4DGj5ws)]
- launch.log file: [[launch.log](https://salbox.synology.me/drive/d/s/11MpodbfHdckKFblZgNZsu6Yqwx9FrCy/D3t_suCTMxSxZG6EVqV93U5JRj5Jrdzj-xrDAEzyj5ws)]

Could anyone please check this?

Thanks!

### Expected behavior

I hope the vehicle to drive uphill well, and reach to the goal point.

### Actual behavior

But the vehicle could not reach to the goal, as it stucks before the uphill.

### Steps to reproduce

1. Pull image from `https://github.com/autowarefoundation/autoware/pkgs/container/autoware/321564631?tag=universe-devel-cuda-0.39.1-amd64` and run Autoware with Carla, using [autoware_carla_interface](https://github.com/autowarefoundation/autoware.universe/tree/0.39.0/simulator/autoware_carla_interface).
2. Run ros2bag file above.

### Versions

- OS: Ubuntu 22.04 (Docker), Ubuntu 20.04 (host)
- Autoware: [0.39.1](https://github.com/autowarefoundation/autoware/tree/0.39.1)
- Carla: 0.9.15
- autoware_carla_interface: [link](https://github.com/autowarefoundation/autoware.universe/tree/0.39.0/simulator/autoware_carla_interface)

### Possible causes

I guess this issue is from slope compensation, as the issue also occurs at hill, and is observed from the planning simulation too.

maxime-clem, 2024-12-18:
It looks like the slope where ego stops is too sharp (~0.12 rad) for the default configuration (0.1).
Please try increasing the parameter (e.g., to 0.2 since the map seems to contain slopes of up to 0.2 rad).
https://github.com/autowarefoundation/autoware_launch/blob/main/autoware_launch/config/control/trajectory_follower/longitudinal/pid.param.yaml#L80-L81

Kim-mins, 2024-12-30:
Thank you for the response @maxime-clem!!

I tried the planning simulation with your suggested parameter, and I found that the vehicle completes the driving well! Thanks a lot!

By the way, would you mind sharing the idea for the weird shape of the drivable area in z-axis?
In my opinion, the drivable area of uphill in the image above seems not being captured well.

Thank you!

closed 2025-02-17 by idorobotics