autowarefoundation/autoware_universe #9803: Autoware cannot recover after stopping

Kim-mins, 2024-12-26:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi team,

I'm currently running Autoware with Carla, and I found a situation that the ego vehicle does not recover from stopping.
Here's a video of the situation: [[rviz](https://salbox.synology.me/drive/d/s/11UJimzd2HgdLrmaBSA6VLvo09uoF7Gq/Wv7O54B3L0TUrhXXAD0O7ydoLy3KSL-3-W7DAtHNn7Qs)]

I failed to find why the ego vehicle could not recover from stopping, but I think the ego vehicle stops after the following steps:
(The timestamp is for the video above)
1. [0:10] The NPC vehicle in front keep driving backward slowly, and the ego vehicle drives forward correspondingly.
2. [0:25] A large portion of the pointcloud of the NPC vehicle is filtered as below, for some reasons.
[image]
3. [0:27] After the NPC vehicle go out of the trajectory of the ego vehicle, the ego starts to drive forward.
4. [0:30] The pointcloud of the NPC vehicle starts to be recovered and it makes the ego vehicle misunderstand the NPC is driving forward, even though the NPC is driving backward as before. (I could check that the NPC keeps driving backward at the video [here](https://salbox.synology.me/drive/d/s/11UKF07Jpm1zGh6zb64bXOPjwpOPhJif/b2tnwTzdOCuyddSDbTxLSFleBlA_PBnC-PbYAhR5p7Qs).)
5. The ego vehicle stucks and cannot recover from the stopping.

Here's more information of the situation:
- Starting/Goal pose (from launch.log)
```
initial pose - x: 228.787155, y: -1.961268, z: 0.060946
initial orientation - qx: 0.000741, qy: -0.000043, qz: 0.000490, qw: 1.000000
goal pose - x: 395.959991, y: -105.389999, z: 2.300000
goal orientation - qx: 0.000000, qy: 0.000000, qz: 0.707109, qw: 0.707105
```
- Map: Town01 [[Town01](https://drive.google.com/drive/folders/1bZfKbq6H9dF1DveuW7iIJiJKN0-W5l03)]
- ros2bag file: [[ros2bag](https://salbox.synology.me/drive/d/s/11UJiuCyrHbUEbPkAfjh6qM6eUv7mH1z/OhmstDX2kqmDvdH-uyM6YtJGVoOZGZdz-a7pgylpq7Qs)]
- launch.log file: [[launch.log](https://salbox.synology.me/drive/d/s/11UKF1tsawOinrE5c6EPvVQpmKEtav2B/40bWfmPgKhuhJtIIBLeKw8HlCvnzLsZD-mLogNWpq7Qs)]

### Expected behavior

I expected that the ego vehicle to drive forward after the NPC vehicle drives out of the ego's trajectory.

### Actual behavior

However, the ego vehicle could not start to drive after stopping.

### Steps to reproduce

1. Pull image from `https://github.com/autowarefoundation/autoware/pkgs/container/autoware/321564631?tag=universe-devel-cuda-0.39.1-amd64` and run Autoware with Carla, using [autoware_carla_interface](https://github.com/autowarefoundation/autoware.universe/tree/0.39.0/simulator/autoware_carla_interface).
2. Run ros2bag file above.

### Versions

- OS: Ubuntu 22.04 (Docker), Ubuntu 20.04 (host)
- Autoware: [0.39.1](https://github.com/autowarefoundation/autoware/tree/0.39.1)
- Carla: 0.9.15
- autoware_carla_interface: [link](https://github.com/autowarefoundation/autoware.universe/tree/0.39.0/simulator/autoware_carla_interface)

maxime-clem, 2025-02-19:
[image]

Thanks for your bag I started analyzing this issue. 
It looks like the PID controller is incorrectly stuck in the `STOP` state but I have not understood why yet.

maxime-clem, 2025-05-23:
I could not reproduce the issue with the current version of Autoware so it was probably solved. 
Iwill close this issue but let's reopen it if we reproduce the issue again.

https://github.com/user-attachments/assets/0fef6521-4160-4ab9-b041-ec367482fc42

closed 2025-05-23 by maxime-clem