autowarefoundation/autoware_universe #9816: Non-blocking obstacle makes the vehicle stop.

Kim-mins, 2024-12-31:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi team,

I'm currently running Autoware with Carla, and I found a situation that the ego vehicle cannot move due to an obstacle, which does not block the vehicle actually.
Here's the rviz video of the situation: [[rviz](https://salbox.synology.me/drive/d/s/11YNOzNFTSzIxMmuMeiu4ve4mSdMYYHf/YFx9oJYMWd9MLMxFFjhrKZ80kZdA69UL-ersgeO-M8As)]

The vehicle stops twice, and I think the details are as below:
- The vehicle stops at [0:02] first. It seems the pointclouds from the tree are filtered well, but after the vehicle is at below of the tree, the points are not filtered and it makes the vehicle stop.
- After recovering from the first stop ([0:06]), another tree in front makes the vehicle stop. It seems the pointcloud is filtered well at first, but they are not filtered well as the vehicle gets closer to the tree.

I also checked the similar issue https://github.com/autowarefoundation/autoware.universe/issues/6940 and I thought the issue  could be resolved for the latest release, but it seems the issue remains.

Here's more information for the reproduction:
- Starting/Goal pose (from launch.log)
```
initial pose - x: -71.782125, y: 88.019365, z: 0.069840
initial orientation - qx: -0.000343, qy: 0.000011, qz: -0.001882, qw: 0.999998
goal pose - x: -195.079193, y: 29.398745, z: 2.600000
goal orientation - qx: 0.000000, qy: 0.000000, qz: -0.707009, qw: 0.707204
```
- Map: Town05 [[Town05](https://drive.google.com/drive/folders/1y5dYpL6o3dp_nSo79ar-MMwDpvdcQOY8)]
- ros2bag file: [[ros2bag](https://salbox.synology.me/drive/d/s/11YMBYj2kwPjfgS92y2CnRpz1VJynw8S/-RHRokdtEVa4tTv9NmS9bz7hW5uZbfD1-ULlgspCM8As)]
- launch.log: [[launch.log](https://salbox.synology.me/drive/d/s/11YNNYLJ0TEzxCXbQr2ng0XsWHCj9IdQ/44EWYbF_J44mVv9-U3VgRLNk6PFH7p2S-YLvgfOWM8As)]

### Expected behavior

I hope Autoware to ignore the irrelevant pointcloud.

### Actual behavior

However, the irrelevant pointcloud is not filtered and makes the vehicle stop.

### Steps to reproduce

1. Pull image from `https://github.com/autowarefoundation/autoware/pkgs/container/autoware/321564631?tag=universe-devel-cuda-0.39.1-amd64` and run Autoware with Carla, using [autoware_carla_interface](https://github.com/autowarefoundation/autoware.universe/tree/0.39.0/simulator/autoware_carla_interface).
2. Run ros2bag file linked above.

### Versions

- OS: Ubuntu 22.04 (Docker), Ubuntu 20.04 (host)
- Autoware: [0.39.1](https://github.com/autowarefoundation/autoware/tree/0.39.1)
- Carla: 0.9.15
- autoware_carla_interface: [link](https://github.com/autowarefoundation/autoware.universe/tree/0.39.0/simulator/autoware_carla_interface)

badai-nguyen, 2025-01-10:
@Kim-mins Thank you for your rising issue. 
- The new feature which added by https://github.com/autowarefoundation/autoware.universe/issues/5594 was not tested on simulation environment yet so https://github.com/autowarefoundation/autoware.universe/issues/6940 might need to be recheck again as [my comment here](https://github.com/autowarefoundation/autoware.universe/issues/6940#issuecomment-2581570325).
- BWT could you enable the new filter(image_segmentation based filter) [by this](https://github.com/autowarefoundation/autoware_launch/blob/main/autoware_launch/launch/components/tier4_perception_component.launch.xml#L11) and test autoware in `camera_lidar_fusion mode` to test the filter on the simulator simulation?

Kim-mins, 2025-02-27:
I'm really sorry for late reply @badai-nguyen.

Thank you for the response!
I just checked that after your suggested modification, there's no stop and the vehicle successfully reaches to the destination!

Thank you! I'll close this issue soon.

closed 2025-02-27 by Kim-mins