autowarefoundation/autoware_universe #9826: I have found a bug in line 352 in behavior_pah_planner_node.cpp

zhmman, 2025-01-05:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

[image]

### Expected behavior

hi dear:
   when i run planning simulation,the first set goal position, the autoware.universe can router set and run correctly. but when i set positioh secondly, which  router set and run correctly again

### Actual behavior

when i run planning simulation,the first set goal positon, the autoware.universe can router set and run correctly. but when i set positioh secondly, which can not router .

### Steps to reproduce

run planning simulation ,set goal position push auto button. set goal position again.

### Possible causes

i have found in line 352 in behavior_pah_planner_node.cpp
change 
    if (!is_first_time && !has_same_route_id) {   
  to
    if (!is_first_time && has_same_route_id) {

kosuke55, 2025-01-06:
thank you for your report and suggestion.
we can change route with second goal when ego vehicle stops.

[image]
[image]

> run planning simulation ,set goal position push auto button. set goal position again.

you tried to put the second goal while the vehicle is driving?  It is not allowed.
```
[service_log_checker_node-5] [ERROR 1736149893.931478769] [system.service_log_checker]: /api/routing/clear_route: status code 50000 'The route cannot be cleared while it is in use.' (/default_ad_api/helpers/routing_adaptor) (set_error() at /home/kosuke55/pilot-auto/src/autoware/universe/common/autoware_component_interface_tools/src/service_log_checker.cpp:86)
```

putting goal with rviz calls [/api/routing/set_route_points](https://autowarefoundation.github.io/autoware-documentation/main/design/autoware-interfaces/ad-api/list/api/routing/set_route_points/). but it is not allowed to reroute.
we can use [/api/routing/change_route_points](https://autowarefoundation.github.io/autoware-documentation/main/design/autoware-interfaces/ad-api/list/api/routing/change_route_points/) for rerouing while driving.

zhmman, 2025-01-06:
my mean is that: the first time ,i set a goal position and the vehicle arrived the goal position. then i set a new goal position again ,this time the mission planner can not complete router and the Auto button keeps gray.

zhmman, 2025-01-07:
Maybe I didn't speak clearly enough, what I meant was that 
1. i set a goal position and the vehicle arrived the goal position. 
2. then i set a new goal position again ,this time the mission planner can not complete router and the Auto button keeps gray.

 

 
image.png (view on web)
 image.png (view on web)