autowarefoundation/autoware_universe #9844: Erroneous dynamic expansion of drivable area makes the ego vehicle stop.

Kim-mins, 2025-01-07:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi team,

I'm currently running Autoware with Carla, and I found a situation that the ego vehicle stops due to an erroneous dynamic expansion of drivable area.
Here's the video of the situation: [[rviz](https://salbox.synology.me/drive/d/s/11eHR2chyqXx1CsPMgslA59V6zMAVFHg/hHDQPg4Ecus7_DNunw2JNog03mqFQ38o-O7ygUmMk9Qs)]

From the start of the video, a drivable area at the corner sometimes invades the trajectory(I cannot know why the message `outside_drivable_area` does not appear.), and when the vehicle is turning at the corner, it eventually stops as one of the rear wheel is at outside the drivable area, which was caused by the update of the area.

Here's more information of the situation.
- Starting/Goal pose (from `launch.log` file)
```
initial pose - x: 334.279584, y: 168.742911, z: 0.263412
initial orientation - qx: -0.000302, qy: -0.000023, qz: -0.003080, qw: 0.999995
goal pose - x: 212.669266, y: 364.220459, z: 2.857196
goal orientation - qx: 0.000000, qy: 0.000000, qz: -0.003752, qw: 0.999993
```
- Map: [Town04](https://drive.google.com/drive/folders/1ZZYgqQJ8l2VRAZxYVt4shTU_PMzF9C7Y)
- ros2bag file: [[ros2bag](https://salbox.synology.me/drive/d/s/11eHR7OlbW0kyBuUuHbYb2fwNeVIraQh/E3CzYQq85CsYW5eouYbK2GAYIdFOs768-U7NgJBUo9Qs)]
- launch.log file: [[launch.log](https://salbox.synology.me/drive/d/s/11eIIrHK0FrKMjp8PM9s9dBbhR66SbcS/xscUB9d7UyPLX9newHugPZpEMDUFpFEU-d7OACSgo9Qs)]

### Expected behavior

I hope the ego vehicle not to stop before reaching the destination.

### Actual behavior

The vehicle stucks before reaching the destination.

### Steps to reproduce

1. Pull image from `https://github.com/autowarefoundation/autoware/pkgs/container/autoware/321564631?tag=universe-devel-cuda-0.39.1-amd64`, and run Autoware.
2. Run ros2bag file above.

### Versions

- OS: Ubuntu 22.04 (Docker), Ubuntu 20.04 (host)
- Autoware: [0.39.1](https://github.com/autowarefoundation/autoware/tree/0.39.1)
- Carla: 0.9.15
- autoware_carla_interface: [link](https://github.com/autowarefoundation/autoware.universe/tree/0.39.0/simulator/autoware_carla_interface)

maxime-clem, 2025-06-17:
I have been able to reproduce the issue. There is a bug in the dynamic drivable area expansion since the feature is only supposed to push the drivable area bounds futher and not bring them closer to the path.

maxime-clem, 2025-07-14:
The bug happens when the Path goes outside the original drivable area (i.e., the left/right bounds of the lane).
The expansion assumes that the input path is always between the original left/right bounds, but it seems to not always be the case.
The issue may be caused by the input Path being incorrectly resampled, causing its linestring to "cut the corner" and go out of the drivable area. The image bellow shows the input path and drivable area of the expansion, and in Rviz we see the resulting drivable area and the final Path (which has correct points).

[image]

maxime-clem, 2025-07-22:
Fixed with https://github.com/autowarefoundation/autoware_universe/pull/10989

closed 2025-07-22 by maxime-clem